LibreCAD/LibreCAD #474: MText character limit

BrianEV, 2015-01-08:
Not sure what has gone wrong. When I enter 2 lines of text with more than about 30 or 40 characters into the MText window, the text does not appear in the drawing but is replaced with a rectangle outline where the text should be. Shorter text entries work normally. I am running Ubuntu 14.04 and LibreCAD 2.0. I have successfully used this configuration, but on a different computer, so it could be something in my setup, some 'finger problem', or a bug.

BrianEV, 2015-01-14:
This seems to affect only one machine and perhaps is a limitation of 31 characters before the entered text becomes rendered as a rectangle instead of text. Seems to still work OK on my other ubuntu 14.04 machine.

Rallaz, 2015-01-15:
See:
http://forum.librecad.org/only-empty-boxes-frames-are-shown-for-image-text-no-line-thickness-td5706737.html

closed 2015-01-15 by Rallaz

BrianEV, 2015-01-16:
Thank you, Rallaz. You saved me a lot of grief. Simple when you know how!

Do I need to close the issue?