LibreCAD/LibreCAD #477: Trim spline or freehand line

glushko311, 2015-01-18:
For spline and freehand lines command trim isn't working. For arcs and circles are ok, but for spline and freehend lines it will not work.

kromanenko, 2015-01-22:
Confirm this defect for splines, polylines and freehand curve with both "Trim" and "Trim Two" commands.
LibreCAD version 2.0.7, GCC 4.8.2. Qt 5.3.1 (2015-01-03 Release from SourceForge).
OS: Windows 8.1 (64bit)

wolterh, 2015-07-30:
I can confirm this on version 2.0.7-g657f24f.

kludikovsky, 2016-11-06:
Still true with Version 2.1.3.
It is also not possible to trim a splie with the polyline trim function.

As this also affects the results from the plot function, I don't consider this a request, but **a bug**.

robzach, 2016-11-22:
Agreeing with kludikovsky that this seems like a bug rather than a request. I'm having this problem in 2.1.3 as well.

gwhitney, 2019-08-16:
As of master on 2019 Aug 16, if I comment out the condition "&& limitEntity->rtti() != RS2::EntityPolyline" at line 141 of file rs_actionmodifytrim.cpp, in the function RS_ActionModifyTrim::mouseReleaseEvent(QMouseEvent* e), then trimming with a polyLine as the limiting entity appears to work just fine. Is there any reason not to delete this condition specifically blocking polylines? I am happy to supply a pull request which takes the condition out, if it would be helpful.

DuncanLithgow, 2021-01-29:
@lordofbikes would the suggestion by @gwhitney work? Otherwise I can only confirm this limitation is still present in v2.2rc2

danielhjames, 2021-09-02:
Another workaround for trimming polylines on librecad version 2.1.3 in Debian is `Tools -> Polyline -> Delete Node` or `Tools -> Polyline -> Delete between two nodes`.

There is also `Tools -> Polyline -> Trim segments` but it is not clear to me how that relates to `Tools -> Modify -> Trim` and `Tools -> Modify -> Trim Two`.

gwhitney, 2021-09-02:
At this point I have run for two years with the miniscule code change I mentioned above, and I have never encountered any difficulty arising from it. The _only_ reason I have never submitted it as a PR is that nobody more experienced with LibreCAD has ever commented on this point, and simply undoing a check that some previous coder clearly put in explicitly seemed inappropriate given that I have no understanding of why the check was put there in the first place. But as far as I have been able to tell, the check need not be there -- on the other hand, maybe there is some horrendous edge case that I simply have never encountered? If so, then it seems like it might be better to check for the edge case rather than ban trimming with polylines altogether, as it is frequently useful and appears generally safe. But to do that, I'd have to know what could be wrong with it...

lordofbikes, 2021-09-03:
@gwhitney many thanks for your reports.
I can confirm, that your modification basically works, when used with care and when one knows what he's doing.
But I've created polylines during my tests, which didn't end in expected results.
More worse is, that the modification also pertains the _Trim Two_ command. This test ended in an immediate crash for me.

So I'm afraid, that we can't release this simple hack into public. There is much more to do for a safe way to allow polylines as limiting entity. And then there can be still cases, where the intersection for trimming is not absolute defined and produces more than one solution.

Concerning this issue, which is about to trim splines and polylines _(freehand lines are polylines)_, LibreCAD is missing basic functionality in these classes.
When I can imagine a solution for polylines, things are more complicated with splines. I consider to close this and create two separate issues for polyline and spline trimming.

gwhitney, 2021-09-03:
> I consider to close this and create two separate issues for polyline and spline trimming.

That sounds fine to me. I agree, I have no experience or insight into spline trimming, but I use polyline trimming frequently, so it's fine as far as I am concerned to separate them. Also I have a bit of a math background, so if you would like to post or send me a few test cases that lead to problems for Trim and or Trim Two with polylines, I'd be happy to take a look and see if I can suggest well-defined behavior, and hopefully in addition suggest code to implement it.

gwhitney, 2021-09-03:
I guess technically *this* issue, as stated is just for spline or freehand lines. So maybe the solution is to leave this and open another for polylines? I will let you handle that how you see best, and my offer to look at difficult cases stands, of course.

DuncanLithgow, 2021-09-08:
If this 'patch' can work in most of / a lot of cases is it not possible to check if a certain case is possible with a high degree of certainty. If the specific case is 'unsafe' then a message can let the user know what workarounds they have available.

lordofbikes, 2021-09-09:
@gwhitney freehand line is a polyline, it's just another name from the creation tool. You can check this with the List plugin, it shows freehand line as polyline.

Because polyline is just a container of lines and arcs, all the math is already there. The problems occur when iterating over the container to find the intersection point for trimming. Each entity is handled separately with the existing line/arc methods. Also limiting entities work over their boundaries, e.g. when the limiting entity is too short, the entity to trim is shortened/lengthened to the intersection with the imaginary extension of the limiting entity.
For complex polylines (closed, zig-zag or with arcs) the issue arise, that multiple segments can be valid for various trimming points. The challenge then is to estimate the best choice or give the user the chance to select what he intends.
Another possible solution could be to find a way to select only the wanted segment of the polyline, which is not implemented yet.

Concerning this issue, to allow polylines to be trimmed with the Trim tool, similar issues arise. This is too much for now, we'll create a new feature request issue for this and can see there what is to do and which pitfalls will arise.

@DuncanLithgow the patch only allows polylines as limiting entity. But still this can have several issues as stated above, which can annoy users and flood forum and issues with support requests.
Putting efforts in detecting edge cases and stop processing or avoid crashes, can be more challenging than creating a clean solution. The condition in question is presumably already the solution to avoid unexpected behavior and crashes, only the message is missing.

danielhjames, 2021-09-09:
@lordofbikes In librecad version 2.1.3 on Debian the `Tools -> Polyline -> Trim segments` command sounds like it would trim the polyline where it crosses an intersection point, but it removes nodes within the segment instead.

A trim command in which the user manually selects the relevant segment of the polyline and then the intersection point could work, as then the user would identify precisely where the trim is to take place (just like when deleting the line between two nodes).

gwhitney, 2021-09-09:
Thanks for your further description of the issues. My offer remains: I would be happy to tackle the ambiguity issues you describe, if you could attach example(s) of the crash you mention with "Trim Two" and/or cases in which "Trim" did something unexpected/improper with a polyline limiting entity (and the ban on such commented out). In the absence of concrete cases, as you know, it's very hard to analyze/propose remedies that might safely broaden the usefulness of "Trim" and/or "Trim Two." (Don't get these remarks wrong, I am a huge fan of librecad, and would be glad to pitch in on its ongoing improvement.)