LibreCAD/LibreCAD #478: Drawing a circle tangential of two circles brings crash

glushko311, 2015-01-18:
Steps to reproduce:

1.Create two circles.
2.Create a point.
3.Use the tool "Draw a tangential Circle of two circles, passing a given point" 
 for create a circle tangential of two circles and passing the point.
4.If now you press ESC or take any other tool, program will crash.

kromanenko, 2015-01-26:
I tried to reproduce this on LibreCAD version 2.0.2, but did not have any trouble. 
I`ll try this on version 2.0.7 later at home.

glushko311, 2015-01-29:
I tried to reproduce
Environment:
LibreCAD
Version: master
SCM Revision: 2.0.7-33-g9ccfb49
Compiler: GNU GCC 4.8.2
Qt Version: 4.8.6
Compiled on: Jan 28 2015

kromanenko, 2015-01-30:
Just now i crashed LibreCAD by this way. 
I took nighly build version 2.0.7-33-g9ccfb49 for Windows.

r-a-v-a-s, 2015-04-29:
It still crashes. Here is my output from a Qt creator debugging:

```
lib\math\rs_math.cpp : simultaneousQuadraticSolverFull : line 907
-3.19318e-014*y^4 +(2.88652e-011)*y^3+(-9.44779e-009)*y^2+(1.323e-006)*y+(-6.68348e-005)==0
lib\math\rs_math.cpp : simultaneousQuadraticSolverFull : line 964
ret=(157/322.227/0)
(-25.8124/265.621/0)
(49.2019/181.048/0)
(185.984/135.071/0)
 tangent: 0

Invalid parameter passed to C runtime function.
Invalid parameter passed to C runtime function.
terminate called after throwing an instance of 'std::length_error'
  what():  vector::_M_default_append
Debugging has finished
```

dxli, 2015-05-10:
can not reproduce this one.

Is this platform dependent? I'm on Linux (Gentoo)

r-a-v-a-s, 2015-05-10:
I just did some more testing with the latest commit. The problem seems to be leaving the operation. If I press Esc it crashes; If I right click it crashes; however, if I press the back button on the toolbar then it will not crash.

r-a-v-a-s, 2015-05-10:
Something else I noticed is that, in general:
- use a tool
- switch to another tool
- esc or right click will activate the last tool activated, even if you didn't use it

Therefore if I follow the steps in the original post, but switch to the regular circle tool instead of pressing Esc, and create a circle, then right click once, I am back at the problem tool, and if I press right click again it will crash.

r-a-v-a-s, 2015-05-10:
In fact you can just skip all other steps.
- select the tool on the toolbar
- press esc

r-a-v-a-s, 2015-05-13:
So the question seems to be: what is this tool doing differently on exit than the other tools?

r-a-v-a-s, 2015-05-13:
:D Nice... that fixed it!

closed 2015-05-13 by dxli