LibreCAD/LibreCAD #479: Scrolling with mouse doesn't sync with scrollbars.

baranovskiykonstantin, 2015-01-18:
Scrollbars works fine, but when I scroll with mouse roller of the scrollbars doesn't moves. So, for example, when I scroll screen to top with mouse, roller of the scrollbars stay on this position (middle by default) and if I will try to scroll screen back using scrollbars - I can't.

LibreCAD:
Version: master
SCM Revision: 2.0.7
Compiler: GNU GCC 4.9.2
Qt Version: 5.4.0
Compiled on: Jan 18 2015

OS:
ArchLinux
Linux 3.17.7-1-ck x86_64

Installed from AUR 
https://aur.archlinux.org/packages/librecad/

r-a-v-a-s, 2015-08-10:
If the drawing area is infinite, then there are no boundaries for the scrollbar to sync with.
What about a fixed size mode, where the user can declare the size of the drawing area?

baranovskiykonstantin, 2015-08-10:
Paper size is set to A4.

dxli, 2015-08-25:
currently, if you scroll by the scroll bars, the area is bound to the visible area plus buffer with a fixed size. Therefore, scrolling is limited here.

when scrolling with mouse/keyboard, scrolling is unbound, making the sliders of the scroll bars always centered.

Proposal:

make mouse/keyboard scrolling behavior identical to scroll bars: limited with some boundary.

1. The current unlimited scrolling is not useful to me, because I mostly scroll too much, and get lost in a few steps;
2. If a user does want to scroll beyond the limit, zooming out would do it. Just like on google maps, you do want to zoom out first, if panning by large distance is needed;
3. we do want to implement a "viewAt" method to allow centering the drawing area around some coordinates.

dxli, 2015-08-30:
wheel scrolling is limited to scroll bar ranges

When zoomed in, the scrolling range won't cover the whole drawing

need to see whether this limitation is acceptable.

if not, need to revert this commit

atorrey, 2015-08-30:
It sounds to me like this is sort of related to the bug I filed a few days ago about the scroll bars being much smaller than the drawing area...  IMHO the scroll bar ranges should be about the same as the drawing size you'd get if you did a Zoom Auto to get the drawing all on the screen, or maybe a SMALL amount bigger...  Thus the scroll bars would change the percent of the range that they occupied in proportion to how much one is zoomed in / out, and the range would always allow scrolling to any part of the drawing.

Example - the scroll bar in a browser - the range is always the page length, and the bar itself changes size to match the ratio between the page size and the window size - if the page is short, the bar is most of the range, while if it is long, the bar gets very short....  But you can ALWAYS scroll from the top of the page to the bottom (if things aren't broken ;-} )

In addition, at least on my browser, if I change the zoom level, the bars change to match...

It is harder in a CAD program since in theory one has an 'infinite canvas' that can have a drawing of any size on it.  This is why I was suggesting that probably the best behavior would be for the ranges to reset to match the current drawing any time one does an 'Auto Zoom', and never be smaller.  However since one often wants to add area to the drawing - i.e. when adding another part to it - then the ranges should increase to match if one zooms OUT from the Auto Zoom size.  Thus it would be easy to expand the drawing size as needed, but also have an easy recovery if one expands to much....  Sorry, but no idea how to make it do that...

I know that I would see it as a significant bug if I couldn't scroll to any point on a drawing regardless of zoom level...

ART
------------------
Arthur Torrey - <arthur_torrey@comcast.net>
-------------------

https://github.com/LibreCAD/LibreCAD/issues/479#issuecomment-136091542

dxli, 2015-08-30:
yes, if scrolling is limited, some users may find the limitation is too much and unreasonable.

The current situation as I understand it:

1, scrollbar is always with limited range;
2, free panning leads to lost control for myself;
3, I think it's good to always limit the range to the whole drawing + buffer; (this is not the current behavior);
4, need to implement a "Viewat" command, so user can always specify a location by coordinates.

rubdos, 2016-06-18:
> I think it's good to always limit the range to the whole drawing + buffer; (this is not the current behavior)

That would be very useful indeed. If you have a clue at what things I need to look for that, I'll post a pull request.

closed 2016-09-07 by r-a-v-a-s