LibreCAD/LibreCAD #480: Drawing ellips inscribed in a quadrilateral brings crash

glushko311, 2015-01-19:
This bug is not reproduce every time!
I didn't find any regularity in its reproduction.

Steps for reproduce
1.Create rectangle.
2. Use the tool "Draw ellipse  inscribed in a quadrilateral i. e. tangential to four lines" .
3.Mark four lines of rectangle.
4.When you marked last line application will crash.
5.Maybe for reproduction of a bug you have to repeat steps 1-4 five or six times.

[image]

[image]

kromanenko, 2015-01-22:
I have experimented some more with that tool. 

Actually application fails in those conditions: 
- when lines are connected into correct rectangle
- when lines are situated orthogonally to each other 

In following conditions given function works as expected:
- having 4 scattered lines
- in some random quadrilateral with different sides and angles
- in paralelogram

Screenshot attached. 

[image]

Environment:
LibreCAD version 2.0.7, GCC 4.8.2. Qt 5.3.1 (2015-01-03 Release from SourceForge).
OS: Windows 8.1 (64bit)

dxli, 2015-05-13:
Please test with the latest nightly builds, for windows:

https://sourceforge.net/projects/librecad/files/Windows/NightlyBuilds/

r-a-v-a-s, 2015-05-13:
I could not reproduce this crash. One thing I noticed is 
[image]

Is this a case where the ellipse does not exist?

dxli, 2015-05-14:
for a trapezoid, if it's not isosceles, there's no inscribed ellipse.
We may have some tolerance issue in detecting trapezoid though

dxli, 2015-05-14:
sorry, there's no "unique" inscribe ellipse for "non-isosceles trapezoid".
More work is needed to handle "infinite number of inscribed ellipses":

for parallelogram and trapezoid

kromanenko, 2015-05-14:
Verified on Windows nightly build 2.0.7-274-g821f837. Now it works.

dxli, 2015-05-21:
fixed for:

parallelograms (and rectangles)
trapezoid isosceles
cleaned up for quadrilateral identification

closed 2015-05-21 by dxli