LibreCAD/LibreCAD #481: Hatch for splines isn't working

glushko311, 2015-01-19:
Steps to reproduce:
1.Create a spline.
2.Turn on "Snap on Endpoints".
3.Close a spline.
4.Try to create hatch or fill figure formed by a spline.

[image]

leogargu, 2015-01-20:
@glushko311 I have reproduced the problem you mentioned (hatch area=0). I am looking into it.

leogargu, 2015-01-21:
OK, I believe there are two different things here:
* Applying hatch on a closed spline does not work. However, there is a workaround: exploding the closed spline before hatching. This does work and a non-zero value for the area is returned. I agree that it would be nice for the hatch to work without having to use explode first, but I don't know yet how easy that is to implement.
* Hatching a circle returns area=0. This is a bug. I have opened a new issue (#482) and I am working on it.

kromanenko, 2015-01-22:
>  However, there is a workaround: exploding the closed spline before hatching

I tried to explode opened spline, closed spline, self-intersected spline. But I could not to explode any spline.

kromanenko, 2015-01-26:
Hi! I have some updates on this issue. 
I have installed LibreCAD on my Linux work machine from package repository (Version 2.0.2, GCC 4.8.2. Qt 4.8.6, OS: Ubuntu 14.04 LTS (64bit) )
On this version i can explode splines. 
Therefore i create separate issue for LibreCAD version 2.0.7 for Winodws: https://github.com/LibreCAD/LibreCAD/issues/488

DuncanLithgow, 2021-01-20:
If I understand correctly a patch was pushed by @bert and accepted by @lordofbikes - why is this bug still open?

lordofbikes, 2021-01-21:
The fix is not related here, the no. 481 means [sf.net ticket 481](https://sourceforge.net/p/librecad/bugs/481/). GitHubs auto formatting can't differentiate this.
The issue still exists, the math for splines, specifically finding intersections with splines, is still not implemented yet in LibreCAD.
That's why it is still open and I added the feature request label now.

M15071060, 2022-05-29:
Late ping, I wanted to confirm this also appears to be not functioning on Debian

<pre><b>       _,met$$$$$gg.</b>          <font color="#F66151"><b>m</b></font>@<font color="#F66151"><b>debian</b></font> 
<b>    ,g$$$$$$$$$$$$$$$P.</b>       ------------ 
<b>  ,g$$P&quot;     &quot;&quot;&quot;Y$$.&quot;.</b>        <font color="#F66151"><b>OS</b></font>: Debian GNU/Linux 11 (bullseye) x86_64 
<b> ,$$P&apos;              `$$$.</b>     <font color="#F66151"><b>Host</b></font>: Z97X-UD5H 
<b>&apos;,$$P       ,ggs.     `$$b:</b>   <font color="#F66151"><b>Kernel</b></font>: 5.10.0-14-amd64 
<b>`d$$&apos;     ,$P&quot;&apos;   </b><font color="#F66151"><b>.</b></font><b>    $$$</b>    <font color="#F66151"><b>Uptime</b></font>: 6 hours, 57 mins 
<b> $$P      d$&apos;     </b><font color="#F66151"><b>,</b></font><b>    $$P</b>    <font color="#F66151"><b>Packages</b></font>: 2825 (dpkg), 15 (snap) 
<b> $$:      $$.   </b><font color="#F66151"><b>-</b></font><b>    ,d$$&apos;</b>    <font color="#F66151"><b>Shell</b></font>: bash 5.1.4 
<b> $$;      Y$b._   _,d$P&apos;</b>      <font color="#F66151"><b>Resolution</b></font>: 2560x1440 
<b> Y$$.    </b><font color="#F66151"><b>`.</b></font><b>`&quot;Y$$$$P&quot;&apos;</b>         <font color="#F66151"><b>DE</b></font>: GNOME 3.38.6 
<b> `$$b      </b><font color="#F66151"><b>&quot;-.__</b></font>              <font color="#F66151"><b>WM</b></font>: Mutter 
<b>  `Y$$</b>                        <font color="#F66151"><b>WM Theme</b></font>: Adwaita 
<b>   `Y$$.</b>                      <font color="#F66151"><b>Theme</b></font>: Adwaita-dark [GTK2/3] 
<b>     `$$b.</b>                    <font color="#F66151"><b>Icons</b></font>: Boston [GTK2/3] 
<b>       `Y$$b.</b>                 <font color="#F66151"><b>Terminal</b></font>: gnome-terminal 
<b>          `&quot;Y$b._</b>             <font color="#F66151"><b>CPU</b></font>: Intel i7-4790K (8) @ 4.400GHz 
<b>              `&quot;&quot;&quot;</b>            <font color="#F66151"><b>GPU</b></font>: AMD ATI Radeon RX 470/480/570/570X/580/580X/ 
                              <font color="#F66151"><b>GPU</b></font>: Intel HD Graphics 
                              <font color="#F66151"><b>Memory</b></font>: 6017MiB / 31987MiB 
</pre>

LbreCAD
Version: 2.1.3
Compiler: GNU GCC 10.2.1
Compiled on: Jan 30 2022
Qt Version: 5.15.2
Boost Version: 1.74.0
System: Debian GNU/Linux 11 (bullseye)

u2fly, 2025-04-18:
>The issue still exists, the math for splines, specifically finding intersections with splines, is still not implemented yet in LibreCAD.

Version: 2.2.2.3-alpha<br/>Compiler: GNU GCC 13.3.0<br/>Compiled on: Apr 18 2025<br/>Qt Version: 6.4.2<br/>Boost Version: 1.83.0
System: Debian GNU/Linux 12 (bookworm)