LibreCAD/LibreCAD #483: Implement tiled printing

melnikov-ivan, 2015-01-22:
It's a feature request.
I use LibreCAD to draw simple drafts, only lines and arcs. And I wish I could print it in real scale on several pages using casual A4 printer and glue sheets together. In print preview mode I could select page number and move printing region. 
Most of people don't have those large printers, but they do want to use CAD software. LibreCAD is great ;)

kromanenko, 2015-01-23:
This`s lovely feature of KOMPAS-Graphic :) Most commercial CADs do not have it.

ivanmihalich, 2015-10-11:
Really this is a very handy feature!
All of the above, in my opinion, should be added:
1. Fitting to the format, when the drawing of the big format (A2, A1 etc) can be printed on a small sheet of A4 or A3. Conversely, the small print of drawing on large format.
2. The rotation of the drawing on the sheet. 
3. The function "include/exclude a page for printing". When the drawing of large format (A1, A0) is printed on several A4 in order not to overflow the printer memory, you must add the "include/exclude a page for printing". The drawing will be printed parts. This function is only in kompas-grafik.
http://kompas.ru/kompas-grafik/about/#about
Best regards!

PR #1091 (2019-05-12): Margins and Tiled Printing
fixes #428
fixes #483
closed 2019-05-12 by lordofbikes