LibreCAD/LibreCAD #484: The "Snap Intersection" does not work for large coordinates

rozos, 2015-01-23:
### Example 1, "Snap Intersection" works OK.
    Circle at 100,100 with radius 30.
    Circle at 120,100 with radius 30.

The "Snap Intersection" works fine at the two intersections of the circles' circumferences.

### Example 2, "Snap Intersection" does not work!
    Circle at 254741, 4286145 with radius 30
    Circle at 254721, 4286145 with radius 30    

The "Snap Intersection" does not work!
- - -

Note: This is a very serious issue in topographical drawings, where large coordinates are common and circles' intersections are used to identify loci.

r-a-v-a-s, 2015-05-13:
This bug exists with the latest commit.

dxli, 2015-05-31:
this is due to precision of equation solvers.

dxli, 2015-08-30:
temporary solution, use ArcArc intersection, instead of quadratic based.

a real solution can be following the KDE/kig project: to have a new quadratic-quadratic solver:

1, find critical lines, straight lines passing intersection;
2, solver linear-quadratic

dxli, 2015-10-06:
a short description of a critical line based quadratic-quadratic solver:

*  identify degenerated quadratic form $$A x^2 + B xy +C y^2 + D x + E y + F =0$$
condition $$Det}| {A B/2 D/2}  {B/2 C E/2} {D/2 E/2 F}| = 0$$

* Degenerated: delta = B^2 - 4 AC
delta > 0: degenerated ellipse
delta= 0: degenerated parabola
delta < 0: degenerated hyperbola

* find a degenerated quadratic by linear combination of two given quadratic (if not degenerated yet), this involves solving a cubic equation.

fa201, 2020-04-13:
@lordofbikes 
I could not reproduce it with this build.
Version: 2.2.0-rc1-190-g51b21cc7
Compiler: GNU GCC 7.5.0
Compiled on: Mar 29 2020
Qt Version: 5.9.5
Boost Version: 1.65.1
System: Ubuntu 18.04.4 LTS

DuncanLithgow, 2021-01-29:
I agree with @fa201 this looks like it's fixed
Version: 2.2.0-rc2
Compiler: GNU GCC 7.3.0
Compiled on: Dec 31 2020
Qt Version: 5.12.4
Boost Version: 1.65.1
System: Windows 10 (10.0)

closed 2021-01-29 by DuncanLithgow