LibreCAD/LibreCAD #486: Insert Image default filetype filter

kromanenko, 2015-01-26:
By default LibreCAD suggests to insert BMP files.
This makes some inconvenience for user, because most images are not stored in BMP format. 

Steps to reproduce: 
1) Click menu "Draw"
2) Click "Insert Images"
3) Have "Open Image" window

Actual: By default "Files of type" filter is set on option "bmp". 
Expected: to have filter by default on "All Image files" position.

Environment:
LibreCAD version 2.0.7, GCC 4.8.2. Qt 5.3.1 (2015-01-03 Release from SourceForge).
OS: Windows 8.1 (64bit)
LibreCAD version 2.0.2, GCC 4.8.2. Qt 4.8.6, OS: Ubuntu 14.04 LTS (64bit)

lordofbikes, 2015-01-26:
LibreCAD should save the last used filter in platform dependent location (Windows: Registry, Linux: hidden config file in home path) when the import is not cancelled. So after selecting once the proper filter and insert an image, the same filter should be selected the next time.

The default for first time, when there is no filter stored, is coded to PNG files, so I assume, that the last time you imported an image you used the BMP filter. 
Please try to change filter to "All Image Files" and then really import an image to a drawing. The next time you open import, "All Image Files" should be selected.

If this isn't the case, there is possibly a permission problem, that the filter setting couldn't be saved.

kromanenko, 2015-01-26:
Some updates per this issue: 
1) For the first time it persistently uses BMP filter on cleanly installed LibreCAD.
2) LibreCAD saves the last used filter successfully. So current issue is not a big trouble.

I have stable reproduction of this behavior on both binary distributions for Windows and Linux, and on fresh compiled from sources: commit a09193a859b5a597a033331a7747470228cc734a, origin/master.

lordofbikes, 2015-01-27:
Well, it was a typo, a needless blank in the filter names why PNG default doesn't work.
I fixed it and set All Image Files as default.

See https://github.com/LibreCAD/LibreCAD/commit/3635091d4cfedf8225025f1b4cdb9bbdf4b761b9

closed 2015-01-27 by lordofbikes

kromanenko, 2015-01-28:
Verified fix on commit 3635091d4cfedf8225025f1b4cdb9bbdf4b761b9