LibreCAD/LibreCAD #487: No tooltip for green hand icon in status bar

kromanenko, 2015-01-26:
There is no tooltip for green hand icon in statusbar. 
I tried to figure out what green hand icon in status bar does. But i still have no idea about that button. 
[image]

lordofbikes, 2015-01-26:
This Icon is part of the Mouse Widget. It has no function. But in drawing mode you can see on left/right side of it the current action for left/right mouse button.

We should apply another icon for this button and possibly select another control instead of a clickable button.

lordofbikes, 2015-01-27:
I fixed this with commit https://github.com/LibreCAD/LibreCAD/commit/9ccfb496037fd1e04755becee16e673c65cd4d2d
It's not longer a button and it has a tool tip now.

closed 2015-01-27 by lordofbikes

kromanenko, 2015-01-28:
Nice mouse icon :)