LibreCAD/LibreCAD #488: Can not explode interpolated splines

kromanenko, 2015-01-26:
Try to apply "Explode" command on following figures:

- interpolated opened spline, 
- interpolated closed spline, 
- interpolated self-intersecting spline. 

I could not to explode any of interpolated splines.

Environment:
LibreCAD version 2.0.7, GCC 4.8.2. Qt 5.3.1 (2015-01-03 Release from SourceForge).
OS: Windows 8.1 (64bit)

kromanenko, 2015-01-27:
Still can not explode interpolated splines on fresh upstream code:
Version: master
SCM Revision: 2.0.7-29-gc27a403
Compiler: GNU GCC 4.8.2
Qt Version: 4.8.6
Compiled on: Jan 27 2015

lordofbikes, 2015-01-29:
Please read https://github.com/LibreCAD/LibreCAD/issues/492#issuecomment-72026229, it takes effect for splines too

dxli, 2015-06-27:
the legacy rs_spline was implemented as a collection of line segment(not based on polyline, but similar).

the new lc_splinepoints was implemented as a single primitive entity, like rs_ellipse. Therefore, tangent of spline is supported.

r-a-v-a-s, 2015-08-10:
Do you want the ability to explode splines?
What is the use case?

dxli, 2015-08-26:
The old rs_spline was implemented as line segments, and can be exploded into rs_lines.

The new lc_splinepoints was implemented as primitives, just like rs_circle or rs_ellipse, therefore, can not be exploded in the same sense as rs_spline.

since splinepoints is a prmitive type, tangent was implemented, while we don't have tangent for rs_spline (a god new feature to implement, finding the stationary points in term of relative angle to a given point through all line segments. An O(N) algorithm).

jack-sprat, 2018-02-20:
@r-a-v-a-s : the use case is to do anything else with the spline! Until it is exploded it's not possible to snap to endpoints, or to create a closed shape, so no hatching is possible. It appears that once created, it is impossible to alter the number of nodes, so for complex shapes, if you don't get it right first time, the choice is either start again from scratch, or explode, (convert to polyline), and modify the elements.

DuncanLithgow, 2021-01-29:
As noted there is no plan to implement exploding splines, nor much logic in trying as per #492 . Feel free to suggest other functionality that would solve similar tasks.

closed 2021-01-29 by DuncanLithgow

u2fly, 2025-04-22:
- https://forum.librecad.org/Spline-through-points-poly-line-td5726973.html#a5726975
> Exploding the quadratic bezer is not currently supported

@dxli 

"Spline through points" could be exploded to polyline with straight segments
- Spline > Explode spline to lines > [v] To Polyline