LibreCAD/LibreCAD #489: Interpolated spline is unknown entity

kromanenko, 2015-01-27:
1) Load LibreCAD
2) Draw Interpolated Spline
3) Go to menu: Info -> List Entities
4) Select Interpolated Spline

Actual:
```n 1: UNKNOWN
  Layer: 0
  Color: #000000
  Line type: SolidLine
  Line thickness: 0.00mm
  ID: 103```
![screenshot from 2015-01-27 17 18 12](https://cloud.githubusercontent.com/assets/10499375/5921827/63674ff0-a64f-11e4-9aac-89d27ce0ab04.png)

Expected: spline, interpolated spline.

Version: master
SCM Revision: 2.0.7-29-gc27a403
Compiler: GNU GCC 4.8.2
Qt Version: 4.8.6
Compiled on: Jan 27 2015

glushko311, 2015-01-29:
I reproduce it with environment:
OS: Windows 7 x64
LibreCAD 2.0.7

OS: Ubuntu 14.04
LibreCAD 2.0.7-33-g9ccfb49 compiled on Jan 28 2015

lordofbikes, 2015-01-29:
It seems that with implementation of splines the "List entities" plugin was not updated. So it has to be implemented as it is unprecedented.

dxli, 2015-08-25:
fixed by commit: 757c3c124abe0c607d9a991b52219a6c69dbb91d

closed 2015-08-25 by dxli