LibreCAD/LibreCAD #490: Ellipse: possible issues with new created ellipse when MinorRatio > 1

lordofbikes, 2015-01-27:
* Create an ellipse where the major radius (1st click) is smaller than the minor radius (2nd click).
* Select the ellipse
* see, that there are no foci marked, only the center

On an ellipse where MinorRatio is < 1, the foci are marked when it is selected.

Problem is, that the equation solver returns no result for getFoci() method in this case.
This may be the source for other issues with ellipse too.

Saving the file and reopen it solves the problem, because somewhere the ellipse is nomalized, so MinorRatio becomes < 1.

I think we have to check on creation and modification of an ellipse when the MinorRation becomes > 1 and normalize it.

r-a-v-a-s, 2015-05-13:
This appears to have been fixed.

kromanenko, 2015-05-14:
Really, now foci appeared for both types of ellipses. And I have noticed one strange detail. 
On ellipse with MinorRatio < 1 i can easily change ellipse by dragging foci by mouse on any small distance.
But on ellipse with MinorRatio > 1 it`s hard to drag foci. Dragging begins to work only when i move foci for relatively large distance (almost as large as major radius).

lordofbikes, 2015-05-14:
Your observation is correct, it seems that the solution for this issue is not clean.
While the MinorRatio is >1 there is no reaction on dragging the foci.
When the MinorRatio becomes >1 while dragging, the ellipse is modified.

I think we are only clean when the ellipse is normalized on creation in sense of DXF definition.
That means the endpoint of MajorAxis must be swapped from 1st click to 2nd click, this will change start and end angle by + or - 90° and the MinorRatio will become <1.
Then all methods handling ellipses should work fine.

r-a-v-a-s, 2015-08-10:
黎東旭 worked on this. Please see if the behavior is as you would expect.

r-a-v-a-s, 2015-09-09:
:bell: I'm going to close this tomorrow if there is no feedback... it seems to be fixed.

closed 2015-09-10 by r-a-v-a-s