LibreCAD/LibreCAD #492: Can't explode circles or ellipses

glushko311, 2015-01-29:
Environment:

OS: Windows 7 x64
LibreCAD 2.0.7

Steps to reproduce:

1.Draw circle or ellipse
2.Use tool explode
3.Mark circle(ellipse) and push Enter

[image]

glushko311, 2015-01-29:
This bug reproduced with environment:
OS:Ubuntu 14.04
LibreCAD 2.0.7

kromanenko, 2015-01-29:
By the way, as far as i remember, AutoCAD can not explode circles, but it can convert ellipses into polylines. But I may be wrong here :)

lordofbikes, 2015-01-29:
No, you are right.

Explode tool is to split entity container. A block or a polyline are entity container, they are composed from several single entities (line, arc, circle) which are grouped together. With explode the connection between the single entities is cut.
As circle and ellipse already are single entities, they can't be exploded in the sense of the tool.
Thus there is no implementation of explode tool in circle or ellipse classes.
You can use Modify -> Divide tool to achieve this.

But I think I know what you want. For machining purpose, it's necessary to convert a circle in multiple arcs or create an ellipse approximation with circular arcs to generate tool paths.
Therefore a simpler tool may be useful, possibly as a part of a CAM module. Maybe we can create a plugin for the beginning.

I was thinking about implementing explode method for circle and ellipse classes, but I think for a good solution both conversions need some parameters and the explode tool has no parameters at all.
For converting a circle we may need a starting point or angle and the number of arcs.
For a closed ellipse we may define a starting point too and generally need a max. drift value. As the conversion of an ellipse is only an approximation, the number of arcs depends on the max. drift and the eccentricity of the ellipse.

All this extends to splines too.

kromanenko, 2015-01-30:
By the way, what`s a point of non-interpolated spline instrument? It looks like polyline with chamfered vertices.

Rallaz, 2015-01-31:
In AutoCAD you can't explode circles, ellipses or splines.
For ellipses and splines you can use the same tip in ACAD or LibreCAD, save as dxf R12.
Since it such entities are unsupported in R12 are converted as polylines

dxli, 2015-09-05:
the old spline is a leftover.

maybe, we can remove it.