LibreCAD/LibreCAD #493: Doubled menu items

kromanenko, 2015-01-30:
Following menu items are doubled:
"Align" and "Same properties" in *Modify* menu.
"List entities" in *Info* menu.
"Sample plugin" in *Help* menu.

Version: master
SCM Revision: 2.0.7-34-gb6b97ea
Compiler: GNU GCC 4.8.2
Qt Version: 4.8.6

korhadris, 2015-03-27:
Do you have both a source-based build and an installed build?

I get this same issue and it is because it is loading plugins from /usr/share/librecad/plugins (installed by Ubuntu package) as well as my current development build in a local area (e.g.  /builds/LibreCAD/unix/resources/plugins). This same problem would occur if you have two sets of plugins in some of the directories it searches to find plugins in general. For fun, I checked this by creating a symbolic link in my $HOME/.librecad/plugins that points to one of the other plugins directories. Now, I get triplicates :)

We could change it so that it will not load a plugin by the same name if one has already been found. There will still be a potential problem with the order in which plugins are found and loaded. In my case, I am getting the /usr/share plugins loaded first, then it is loading my development plugins.

kromanenko, 2015-03-27:
Yes, i have both: source-based build and package-installed build.

lordofbikes, 2015-03-30:
fixed by korhadris with commit d464a9354e32f99e62b520d7e4c2842c8c9315e7

closed 2015-03-30 by lordofbikes