LibreCAD/LibreCAD #495: Commit number in revision info

kromanenko, 2015-01-30:
First i pulled latest code from master. 
Head commit was 
```
commit b5075508fdff1282249c55be6857fe633789aba4
Author: Dongxu Li <dongxuli2011@gmail.com>
Date:   Wed Jan 28 23:04:52 2015 -0500
```
Then i have compiled LibreCAD and viewed version on Help -> About
I see there SCM Revision: 2.0.7-34-**gb6b97ea**.

I found one of previous commits: b6b97ea, but it`s name matches not exactly.

Screenshot attached.
[image]

dxli, 2015-05-30:
the commit number is auto from:
git describe --tags

lordofbikes, 2015-05-30:
This is a duplicate of #494, closing it

closed 2015-05-30 by lordofbikes