LibreCAD/LibreCAD #496: File operations and zooming icons on Linux

kromanenko, 2015-01-30:
Windows build of LibreCAD uses it`s own icons as expected. 
LibreCAD on Linux takes file operations and zooming icons from system libraries. 
Screenshot to compare:
[image]

atorrey, 2015-01-30:
I might get others that disagree, but IMHO using the system library icons is a GOOD THING! I already know what the system icons are, and what they mean / do, so unless the LibreCAD use does something other than what the icon normally does, using the standard icons means I don't have to learn yet another set of icons for the same task...

Now if the icon does something non-standard that the user might not expect, then having a different icon makes sense, but for basic tasks like opening / closing files, it doesn't change, so why do we need a different icon...

This is especially useful for someone that isn't using the program daily, as it cuts the fumbling time to remember the interface details....

ART

------------------
Arthur Torrey - <arthur_torrey@comcast.net>
-------------------

https://github.com/LibreCAD/LibreCAD/issues/496

kromanenko, 2015-02-02:
I agree with you that it`s good to use system icons. Furthermore Gnome and KDE icons have very good look-and-feel. Unfortunately there are only common UI icons, without special CAD-related ones. 
And mix of several icon-sets in one program looks not good.

atorrey, 2015-02-03:
Mixing isn't a big deal in my opinion, I sort of expect it if a program is going to be doing both stuff that fits in the usual file type stuff, and other things...

As long as the program specific icons are about the same size and don't look TO drastically different, not a big deal.  A bit of mismatch doesn't hurt as it helps distinguish the special icons from the regular ones...

ART

------------------
Arthur Torrey - <arthur_torrey@comcast.net>
-------------------

https://github.com/LibreCAD/LibreCAD/issues/496#issuecomment-72446615

r-a-v-a-s, 2015-08-10:
My vote is to have LibreCAD always use its own icons. 
More people are using super HD displays; and we will be implementing a custom icon size option.
I think it is smart for us to know how the icons will be scaling.
I also think that consistency within the program takes priority over 
users on certain operating systems seeing icons they *might* be familiar with.

dxli, 2015-08-25:
From the user perspective, native icons could be more intuitive.

It would be interesting to use our own SVG icons for HD displays. More GUI work for HD.

r-a-v-a-s, 2016-06-02:
I've allowed users to choose to use themed icons;
it's off by default for consistency;
and also because the mixing looks awkward on some desktop environments.

closed 2016-06-04 by r-a-v-a-s