LibreCAD/LibreCAD #497: Two problems with drawing a circle tangential to three entities

robnee, 2015-01-31:
I am trying the draw circles that are tangential to two concentric circles and a line that is radial to the circle's common center point.  I have run into two problems

1) It is very difficult to select all three entities.  Often, LibreCAD will refuse to allow me to select the third entity.  It's as it it ignore clicks on entities is views as invalid or for which it assumes or predicts a solution cannot be found.  This if frustrating.  Sometimes if I select the entities in varying orders it will work.  For example, inner circle then outer circle then line.  Sometimes I must do line first then inner circle (never outer circle) then outer circle.  Sometimes no combination will work.  Sometimes changing the snapping options and zoom helps but not in any predictable way that I can determine.

2) Once I have the two circles and a line finally selected LibreCAD does not seem to compute and offer all possible solutions.  In this case I believe there should be 4 but only 2 are shown.  See attached image for an example.

I suspect that part of the problem is that the circles are concentric.  I have noticed a surprising amount of 'ingenuity' in the solutions that LibreCAD calculates for a given set of entities so it looks to me from the outside that the code is very well done.  I'm wondering if this is some sort of edge case and therefore fairly straight-forward to fix.  I hope I've been clear in my descriptions.

[image]

r-a-v-a-s, 2015-04-30:
[image]

**first case**

* create a circle centered at 0,0 with radius 100
* create a circle centered at 100,0 with radius 100
* create a triangle based on their intersections and point -100,0
* activate the Tangential 3 Circles tool
* select the side of the triangle that connects the top intersection and point -100,0
* select the side below it
* select the vertical side

Only the incircle is given as an option. Select the sides in a different order to receive all options.

**second case**

Repeat the same construction as above but start at 200,200 instead of 0,0.
At the step of selecting the vertical line, the program fails to select the line.
If you instead select the vertical line first it will allow the selection of all three;
however, it will give only the options of an excircle tangent to one of the non-vertical sides.

r-a-v-a-s, 2015-05-21:
This appears to be fixed regarding my two cases.

dxli, 2015-05-29:
I will go through the test cases

dxli, 2015-05-30:
passed all test cases here

r-a-v-a-s, 2015-05-30:
The tool seems to be working as expected on Windows 7. I can not reproduce the original issue now.

closed 2015-05-30 by dxli