LibreCAD/LibreCAD #499: Can`t change color, lineweight, linetype by toolbar

kromanenko, 2015-02-02:
Can`t change color, lineweight, linetype by toolbar "Pen selection" (properties?).
Steps: 
1) Draw some objects
2) Select objects
3) Select new Color, Lineweight, Linetype
Expected: to have changed properties by clicks on new values. 
Actually: objects properties remained intact, verified them visually and by "List entities" tool.
New settings will be applied only to next drawn objects.

[image]

lordofbikes, 2015-02-05:
There is everything OK with the steps you described, that's the way it should work.

To change properties of existing entities use Modify->Properties for single entity or Modify->Attributes for multiple selected entities.

Or use pen "By Layer" and then modify layer attributes to change all entities on that layer.

closed 2015-02-05 by lordofbikes

kromanenko, 2015-02-06:
Hi! I know about that two instruments, but it takes 5 mouseclicks and much eye-searching over screen to change one property of object. 
I expected common CAD experience with toolbar that should take only 2 clicks. It`s very much helpful when working with complicated multi-layered drawings.

glushko311, 2015-02-06:
I have some experience of working in CAD systems. Using  panel for change entity properties more convenient and more clearly then using instruments from menu "modify". And most importantly it's more faster, this feature saves a lot of working time of engineer. 
Changing of entity properties is often used operation, so the simplicity and speed very important here.

vladtcvs, 2020-11-09:
I'm agree with previous 2 people