LibreCAD/LibreCAD #505: command 'mv' -  starts vertical dimension instead of move command

mstroehle, 2015-03-14:
I've read the wiki article "using the command line"
the command 'mv' does not start the move command, it starts the vertical dimension.
changing GUI Language and or command language doesn't take effect.
I hope i have correct posted this Issue.

lordofbikes, 2015-03-18:
The wiki article is consistent with actual source code, I've checked this. Vertical dimension should be 'dv' command.

To solve your issue, its important to know your LibreCAD version and Operating System.
There is a command cache file, where the wrong command possibly was set by command translation or previous version.
The location of the cache file depends on your OS.

kromanenko, 2015-05-13:
On current build (commit 9fb19029be7291bf9a5871f54a2b25a2e9eb0c50) for Linux "mv" command successfully starts "Move" command.

closed 2015-05-13 by lordofbikes