LibreCAD/LibreCAD #506: Please consider enable the library LLGE in the project LibreCAD

ivanmihalich, 2015-03-14:
Hello dear The LibreCAD Team!
I thank You and respect for the tremendous work to support and improve LibreCAD!

Please consider enable the library LLGE (GNU GPL V2) in the project LibreCAD.
Or at least post a link to the project LLGE on the resource savannah.gnu.org
This library contains the graphical notation elements of electric circuits, kinematicheskikh schemes, refer to plank and other.
In my opinion it would be convenient for novice users.

Project references:
https://savannah.gnu.org/projects/llge/
http://git.savannah.gnu.org/cgit/llge.git
http://git.savannah.gnu.org/cgit/llge.git/snapshot/llge-master.tar.gz

With respect and best wishes,
Ivan M. (Mihalich_) author LLGE.

dxli, 2015-03-27:
Thanks!

We will do that, and may ask questions in the process!

ivanmihalich, 2015-04-15:
Thank you very much !

closed 2015-04-24 by dxli

Symbian9, 2023-02-27:
Due to `../library/sheets/` in whole or in part made with QCAD CE, there some issues with fonts (for example, see `../A3_2_draw.dxf`):

- https://github.com/LibreCAD/LibreCAD/tree/master/librecad/support/library/sheets

Guess, it needs to be fully redesigned. Standard: *GOST 2.301-68 (ESKD)*

[image]