LibreCAD/LibreCAD #512: Adding support for osifont

triplus, 2015-04-09:
Hi.

I explored a bit about the situation regarding fonts and LibreCAD. On LibreCAD Wiki i have seen the future plan for adding support for Qt fonts and on forum i have seen instructions on how to convert fonts to font format LibreCAD uses ATM.

It was mentioned a few times on the forum you are always looking for new fonts with appropriate licences to be included in LibreCAD by default. As i use osifont with FreeCAD and it delivers i was wondering if LibreCAD would be interested in including the osifont font by default.

https://code.google.com/p/osifont/

dxli, 2015-04-24:
It's by GPLv3.

Is it compatible with GPLv2?

triplus, 2015-04-24:
Good question. We could ask FSF and the original author on his thoughts. I am not a lawyer but this is my understanding of the issue.

LibreCAD licensed under GPLv2 would use OsiFont licensed under GNU GPL license version 3 with GPL font exception. Converting this font to stick font format would produce derivative work font licensed under GNU GPL license version 3 with GPL font exception. It is important to state GPL font exception was preserved.

That would make directly incorporating the GPLv3 font in GPLv2 software not possible. 

https://www.gnu.org/licenses/gpl-faq.html#AllCompatibility

But there is always possibility to crate an "bundle" or "aggregate". Basically at install time user would need to be notified OsiFont is a product separate from LibreCAD, what it is and where it came from,
that they can obtain it themselves for free, and that they can freely distribute it themselves.

http://stackoverflow.com/questions/5176926/gnu-freefont-gpl-licence
https://www.gnu.org/software/freefont/FAQ.html

I am nut sure if its worth the hassle and i will let developers of LibreCAD to decide. I wouldn't want any legal issue whatsoever be the outcome but based on my understanding it could be done by distributing OsiFont with LibreCAD as aggregate and not by simply incorporating the OsiFont in LibreCAD.

hikikomori82, 2016-05-12:
Osifont will probably soon be multi-licensed (from GPLv3FE to GPLv3FE + LGPLv3FE), adding one more license (GPLv2FE) is not problem I think, would it solve your issue?