LibreCAD/LibreCAD #513: DWG drawing integrity

kromanenko, 2015-04-10:
I tried to open my drawing of my bike. 
Looks that many curves from MAIN and THIN layers are lost while opening.
This drawing made of lines, arcs, circles and splines.

Actual:
[image]

Expected. I have not AutoCAD now, thus i can not check those two missing arcs on front wheel rim :)
[image]

Drawing is also available: https://drive.google.com/file/d/0B8cQdxA1Np1PR3FoZDhDWm13VTA/view?usp=sharing

Rallaz, 2015-04-10:
To solve this "problem" are required math spline.

Added a hack to solve initialy.
[image]

kromanenko, 2015-04-14:
It works much better. Only small spline in aft lights part remained invisible. 
[image]
Also I tried it in DraftSight for reference. As far as i concerned it has officially licensed DWG support from Autodesk to Dassault.

Rallaz, 2015-04-14:
Really all splines are read ok, but have a difference in dxf to dwg
I think:

Rational spline (nurbs) in Dxf flag 12, and a list of control points
In Dwg flag 12, and a list of knot points and control points
Non-rational (bezier) in Dxf flag 8, and a list of knot points, control points and fit points
In Dwg flag 8, and only a list of fit points

The missing spline have only 3 fit points and require 5 control points to show

To solve are required a function to calculate the five control point using the 3 fit points stored in dwg

DuncanLithgow, 2021-01-30:
I don't see anything for a bug report here. Please re-open this issue if there is a bug in the latest version of LibreCAD. Otherwise this looks like something for the forum.