LibreCAD/LibreCAD #514: Compile issues

PlastecProfiles, 2015-04-10:
I have been trying to compile on windows 7 I just downloaded Qt creater this morning and boost but i keep having these errors. 
[image]
Also how do I do this?"Select librecad as building target in Qt Creator (instead of tff2lff, which is another choice)"

PlastecProfiles, 2015-04-10:
I forgot to mention that i used the Offline installer with MuParser.

dxli, 2015-04-11:
looks like the file muparser.pro need a fix

PlastecProfiles, 2015-04-13:
Do you have a fix?

dxli, 2015-04-13:
can you have a look at the muparser.pri file within the source file folder, looks like it fails to use the muparser source included in LibreCAD

PlastecProfiles, 2015-04-13:
Here is the file opened in a text editor. I am new to this so I am not sure if this file is correct or not?
[image]

PlastecProfiles, 2015-04-13:
Just in case you want to look at this file as well here it is.
[image]

kromanenko, 2015-04-14:
Currently i also have problems with compilation in Linux. 
Steps: 
$ qmake-qt4 librecad.pro
$ make
Stderr output listed below: 
```
src/drw_entities.cpp: In member function ‘virtual bool DRW_Image::parseDwg(DRW::Version, dwgBuffer*, duint32)’:
src/drw_entities.cpp:2240:14: warning: unused variable ‘clipMode’ [-Wunused-variable]
         bool clipMode = buf->getBit();
              ^
src/drw_entities.cpp:2234:13: warning: unused variable ‘displayProps’ [-Wunused-variable]
     duint16 displayProps = buf->getBitShort();
             ^
src/drw_objects.cpp: In member function ‘virtual bool DRW_ImageDef::parseDwg(DRW::Version, dwgBuffer*, duint32)’:
src/drw_objects.cpp:1165:15: warning: variable ‘size’ set but not used [-Wunused-but-set-variable]
     DRW_Coord size = buf->get2RawDouble();
               ^
src/intern/dwgbuffer.cpp: In member function ‘double dwgBuffer::getBitDouble()’:
src/intern/dwgbuffer.cpp:358:57: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
         double ret = *reinterpret_cast<double*>( buffer );
                                                         ^
src/intern/dwgbuffer.cpp: In member function ‘double dwgBuffer::getRawDouble()’:
src/intern/dwgbuffer.cpp:424:57: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]
     double nOffset = *reinterpret_cast<double*>( buffer );
                                                         ^
Project MESSAGE: We will be using CPP11 features
Project MESSAGE: Found Boost in /usr
Project MESSAGE: Using Boost includes from /usr/include
Project MESSAGE: Using external muparser
Project ERROR: Package libxml++-2.6 not found
make[2]: *** [Makefile] Error 2
make[1]: *** [sub-src-make_default] Error 2
make: *** [sub-librecad-make_default-ordered] Error 2
```
Also `# make clean` fails. 
```
$ make clean
cd libraries/ && make -f Makefile clean
make[1]: Entering directory `/home/kromanenko/git/LibreCAD/libraries'
cd libdxfrw/ && make -f Makefile clean
make[2]: Entering directory `/home/kromanenko/git/LibreCAD/libraries/libdxfrw'
rm -f ../../generated/lib/libdxfrw/obj/libdxfrw.o ../../generated/lib/libdxfrw/obj/libdwgr.o ../../generated/lib/libdxfrw/obj/drw_header.o ../../generated/lib/libdxfrw/obj/drw_classes.o ../../generated/lib/libdxfrw/obj/drw_entities.o ../../generated/lib/libdxfrw/obj/drw_objects.o ../../generated/lib/libdxfrw/obj/drw_textcodec.o ../../generated/lib/libdxfrw/obj/dxfreader.o ../../generated/lib/libdxfrw/obj/dxfwriter.o ../../generated/lib/libdxfrw/obj/dwgreader.o ../../generated/lib/libdxfrw/obj/dwgbuffer.o ../../generated/lib/libdxfrw/obj/drw_dbg.o ../../generated/lib/libdxfrw/obj/dwgreader21.o ../../generated/lib/libdxfrw/obj/dwgreader18.o ../../generated/lib/libdxfrw/obj/dwgreader15.o ../../generated/lib/libdxfrw/obj/dwgutil.o ../../generated/lib/libdxfrw/obj/rscodec.o ../../generated/lib/libdxfrw/obj/dwgreader27.o ../../generated/lib/libdxfrw/obj/dwgreader24.o
rm -f *~ core *.core
make[2]: Leaving directory `/home/kromanenko/git/LibreCAD/libraries/libdxfrw'
cd jwwlib/ && make -f Makefile clean
make[2]: Entering directory `/home/kromanenko/git/LibreCAD/libraries/jwwlib'
rm -f ../../generated/lib/jwwlib/obj/dl_writer_ascii.o ../../generated/lib/jwwlib/obj/dl_jww.o ../../generated/lib/jwwlib/obj/jwwdoc.o
rm -f *~ core *.core
make[2]: Leaving directory `/home/kromanenko/git/LibreCAD/libraries/jwwlib'
make[1]: Leaving directory `/home/kromanenko/git/LibreCAD/libraries'
cd librecad/ && make -f Makefile clean
make[1]: Entering directory `/home/kromanenko/git/LibreCAD/librecad'
cd src/ && make -f Makefile clean
make[2]: Entering directory `/home/kromanenko/git/LibreCAD/librecad/src'
/usr/lib/x86_64-linux-gnu/qt4/bin/qmake -o Makefile src.pro
Project MESSAGE: We will be using CPP11 features
Project MESSAGE: Found Boost in /usr
Project MESSAGE: Using Boost includes from /usr/include
Project MESSAGE: Using external muparser
Project ERROR: Package libxml++-2.6 not found
make[2]: *** [Makefile] Error 2
make[2]: Leaving directory `/home/kromanenko/git/LibreCAD/librecad/src'
make[1]: *** [sub-src-clean] Error 2
make[1]: Leaving directory `/home/kromanenko/git/LibreCAD/librecad'
make: *** [sub-librecad-clean-ordered] Error 2
```

This occurred on commit 9c857e8b4b91d9ede2f9111c56250883b8293253. 
Linux Ubuntu 14.04 LTS, 64bit.

dxli, 2015-04-14:
sudo apt-get install libxml++2.6-2

kromanenko, 2015-04-16:
Sorry, but this did not helped on my both Ubuntu machines.

lordofbikes, 2015-04-16:
You need this too:
`sudo apt-get install libxml++2.6-dev`

kromanenko, 2015-04-16:
Thanks, that works.

closed 2015-04-24 by dxli