LibreCAD/LibreCAD #515: DWG zoom position on opening

kromanenko, 2015-04-14:
When i open my DWG drawing in LibreCAD i can not find where actually my drawing situated on endless model space. 
I have to click "Auto Zoom" button to fit drawing into viewport. 

For example DraftSight and A360 Autodesk online viewer fits opened drawing properly.

This happens on commit 90e2c72 "add hack to pass fit point as control point".

jirislaby, 2016-01-19:
This still happens with 2.0.9. Could it be fixed such that the auto zoom action is performed automatically after the import is finished?

AndreySV, 2016-11-15:
The problem still exist in 

Version: 2.1.2
Compiler: GNU GCC 6.1.1
Compiled on: Sep 17 2016
Qt Version: 5.6.1
Boost Version: 1.61.0
System: Debian GNU/Linux stretch/sid.

I'm agree with jirislaby about executing auto zoom action after import. Because without that it looks like something is broken and import wasn't successful. But the document is loaded.

shawncurry, 2019-05-16:
Bump.  I was just going to suggest this behavior at least be a preference.

As an EU, I would prefer to have my drawing Auto-Zoomed when I open it.

### Expected behavior

When I open my drawing, it is centered and zoomed in the editor.

### Observed behavior

It's not.  I need to click Auto-Zoom most of the time.

### Steps to reproduce or sample file

Open a drawing that is not plotted in the first quadrant.

### Operating System and LibreCAD version info

Win10
LibreCAD 2.1.3

Of course, it begs the question - shouldn't this just be the default behavior?  Or should it be a preference?  Does anybody really want the current behavior?

xanderdin, 2019-05-17:
@shawncurry, Auto-Zoom actually modifies a drawing (telling about DXF, not sure about DWG). And that obviously is not what everyone wants when opening a drawing just for a view. On the other side, I also feel like auto-zooming on opening is a desirable behavior in most cases. Actually, I always Auto-Zoom my DXF drawings before final save and that keeps my drawings fit on next opening, at least with the same LibreCAD window size.

In my opinion, making it a checkbox unset by default in Application Preferences would be a compromise.

lordofbikes, 2019-05-17:
I think DWG in the title means drawing, not DWG format?
There shouldn't be a big difference because DWG and DXF are only different import filters.
But maybe in 2.1.2 DWG support was more rudimentary than today and the import was/is missing VPORT or similar support.
So we should check *libdxfrw* first.

Off the top of my head, the last view position and extends should be stored in the DXF and restored on opening again. This is also the reason why auto zoom marks the drawing modified.
Maybe we have to check DXF references for the correct handling and if *libdxfrw* or LibreCAD is missing something here.

This may be also related to a request I can't quickly find yet. DXF has the possibility to store multiple views to select from. This feature is missing in LibreCAD.

shawncurry, 2019-05-18:
Hi Armin @lordofbikes,

I think @xanderdin is right, if I save the drawing after zooming, it remembers.

I'm changing the tag from import error to a feature request for the preference.  Off by default.  I'll probably update my other pull request when I move the other pref and just add this one while I'm in there.

PR #1122 (2019-06-10): Feature/issue1108
- **Fix #1108 Improve User Experience while working with multiple drawings**
- Fix #515 DWG zoom position on opening
- Fix #1085 Zoom panning can only be used once and must then be react
- Add File > Save All and File > Close All actions + icons
- This commit also includes my un-merged changes from #1099 (preferences stuff)
commit 52652be8 (2019-06-10): - Fix #1108 Improve User Experience while working with multiple drawings
- Fix #515 DWG zoom position on opening
closed 2019-06-10 by lordofbikes