LibreCAD/LibreCAD #516: libxml++2.6 dependency

lordofbikes, 2015-04-15:
I've read up on libxml++2.6 for Windows now and I'm very unlucky with that dependency.
It seems not impossible to get it up and running, but I see serious issues arise in future to support Windows users setting up all dependencies libxml++2.6 needs.
libxml++2.6 is dated around 2004 and I found hints from 2012 to use libxml++1 for Windows because of lesser dependencies.
Another point is about nearly 2MB binary code size expected for the release library, that looks very oversized for me.

Thus, instead fiddling around to fulfil all dependencies, get it running, write a detailed wiki article and support future issues concerning this setup,  I would prefer to invest my time in refactoring rs_makercamsvg.cpp and use another xml writer. 
Aside from that I expect that refactoring will be much cheaper than setup and support for libxml++2.6.

Two candidates from my point of view are:
- boost/property_tree
- QXmlStreamWriter

Both are available for LibreCAD without additional dependencies, they will fix the broken Windows build and I assume that they are much cheaper in terms of additional code size and therefore download volume.

Any other thoughts or XML writer candidates?

dinkel, 2015-04-17:
I rewrote the XML writer by first extracting to a writer interface and then switched to QXmlStreamWriter, so the libxml++ dependency is removed again.

Please check it out here: https://github.com/dinkel/LibreCAD/tree/bugfix-516

@dxli: Could you have a look at it, too?

dxli, 2015-04-18:
it's wonderful!

the RS_XMLWriterQXmlStreamWriter abstraction layer even allows easy switching to other XML writer.

Just one suggestion: we would like to see new additions flagged as "LC_" instead of "RS_", to clarify the LibreCAD additions, could be relicensed, while RS_ legacy code is locked to GPLv2.

Please send in a pull request. You should have direct commit access to the official repository as well, to speed up the process.

Thanks!

dxli, 2015-04-18:
Hi,

I added your commit to the master branch, and renamed files here: 9d0b2f986

I also use unique_ptr to hold writer pointers, and try to use forward declaration as much as possible.

Sorry for being messy :(

dinkel, 2015-04-19:
Cool! I've learned some new stuff by reading through your changes, thanks!

Next time I'm quicker with sending a pull request ;-)

closed 2015-04-24 by dxli