LibreCAD/LibreCAD #517: RS_Arc::draw Segmentation Fault

Palmitoxico, 2015-04-18:
I've recently compiled the abbce43cc8f186d35a6b61dbe379e55f0e7b5840 revision and found a bug when attempting to open an dxf generated with a previous version of Librecad (https://aur.archlinux.org/packages/librecad/), the program crashes immediately.

I've debugged this problem and found the exact part of the code that was triggering this issue:

https://github.com/LibreCAD/LibreCAD/blob/master/librecad/src/lib/engine/rs_arc.cpp#L938-945
```c++
...
    arc.setPen(getPen());
    arc.setSelected(isSelected());
    arc.setReversed(false);
	for(size_t i=0;i<crossPoints.size()-1;i+=2){
        arc.setAngle1(baseAngle+crossPoints[i]);
        arc.setAngle2(baseAngle+crossPoints[i+1]);
        arc.drawVisible(painter,view,patternOffset);
    }
...
```

It seems that sometimes the ```crossPoints``` vector has 0 size and ```crossPoints.size() - 1``` will cause an (unsigned) integer underflow so ```crossPoints[i]``` will be requested even if crossPoints is empty.

I've managed to fix it through the simple, stupid way:

```c++
--- a/librecad/src/lib/engine/rs_arc.cpp
+++ b/librecad/src/lib/engine/rs_arc.cpp
@@ -938,7 +938,9 @@ void RS_Arc::draw(RS_Painter* painter, RS_GraphicView* view,
     arc.setPen(getPen());
     arc.setSelected(isSelected());
     arc.setReversed(false);
-       for(size_t i=0;i<crossPoints.size()-1;i+=2){
+    unsigned int SizeCross = crossPoints.size();
+    if(SizeCross) SizeCross--;
+       for(size_t i=0;i<SizeCross;i+=2){
         arc.setAngle1(baseAngle+crossPoints[i]);
         arc.setAngle2(baseAngle+crossPoints[i+1]);
         arc.drawVisible(painter,view,patternOffset);
```

I don't know if this is the right way to fix it, but now it seems work without problems.

---System---
OS: Arch Linux (x86_64)
Kernel: 3.19.3-3-ARCH
Video Card: Radeon 6850 (Gallium 0.4 on AMD BARTS)

dxli, 2015-04-19:
I feel it's better to do:
for(size_t i=0;i<=crossPoints.size();i+=2){
The bug is due to improper int to size_t switching.

Thanks for reporting!

Can you send in a pull request?

dxli, 2015-04-19:
fixed in commit: e7099ceb1

I am trying to use c++ containers instead of Qt ones, and introduced this bug in process. Sorry for that.

We are going to release 2.1.0-beta soon. Please help testing especially in next weeks.

Since we are going to spend time on GSoC, it would be busy time for us.

Palmitoxico, 2015-04-20:
Ok, I've found some others (unrelated) bugs, and I will fix them when I get some free time.

closed 2015-04-24 by dxli