LibreCAD/LibreCAD #520: LC_SplinePoints::getRefPoints Segmentation Fault

Palmitoxico, 2015-04-21:
The crash always occur when I try to select a curved line.

Function:

```C++
RS_VectorSolutions LC_SplinePoints::getRefPoints()
{
	RS_VectorSolutions ret;

	if(data.cut)
	{
		std::copy(data.controlPoints.begin(), data.controlPoints.end(), ret.begin());
	} else {
		std::copy(data.splinePoints.begin(), data.splinePoints.end(), ret.begin()); //Segmentation Fault
	}

	return ret;
}
```
https://github.com/LibreCAD/LibreCAD/blob/master/librecad/src/lib/engine/lc_splinepoints.cpp#L421

The ```data``` structure uses the ```QList``` class while ```RS_VectorSolutions``` (```ret```) structure uses ```std::vector```, so it seems to cause the crash.

Changing the ```LC_SplinePointsData``` structure to use ```std::vector``` instead ```QList``` requires to modify a lot of code. I'm doing it right now, but will take some time until I finish.

dxli, 2015-04-24:
fixed by d50f25e1

The reason of crash:

std::copy() assumes the target has enough room to hold items, while we have an empty target (ret) here.

using std::back_inserter(ret) solves this issue. std::copy becomes push_back() here.

dxli, 2015-04-24:
thanks for reporting.

Together we can get stabilization faster, and the goal is 2.1.0 release

closed 2015-04-24 by dxli