LibreCAD/LibreCAD #523: snap intersection and incircles

r-a-v-a-s, 2015-04-29:
* create a triangle using the line tool
* create the incircle using geometric construction or the inscribed circle tool
* enable snap intersection
* select the distance point to point tool
* hover the mouse over a point of tangency
* move the mouse side to side in very small movements

You will see that in most cases it is snapping between two points that are extremely close together.

dxli, 2015-04-30:
First, to be fair, a tangential point is not a well-defined intersection point, because it's up to the floating point error.

However, if detected, a tangential point is identified as intersection. I am not clear why there could be two intersection points out of the tangential contact.

Please try to test using "tangential 3", common tangential circle of 3 circles to get inscribed, because a line can be treated as a circle.

Maybe, the tool "inscribed circle" should be removed or linked to "tangential 3"

r-a-v-a-s, 2015-04-30:
I get the same results with that tool, and if it is constructed geometrically. 99.99999693 and 100.00000307 is an example of the difference. I'm aware of the limitation of floating point arithmetic, and thought that might have something to do with it. Keep me updated. I need this solved to be able to use the program; and I am willing to ask around the internet for help.

dxli, 2015-04-30:
can you upload a dxf file to show that?

r-a-v-a-s, 2015-04-30:
I guess we can only upload images here. I sent you the file in an email.

r-a-v-a-s, 2015-04-30:
Note that you won't see the crosshair move. I have my drawing preferences set to decimal and the highest precision. I see the issue in the coordinates at the bottom left of LibreCad.

dxli, 2015-05-01:
I can see the problem. 

We can improve the precision level of equation solvers:

1, need a better algorithm on quadratic-quadratic equation systems (means to find intersections of two quadratic curves); For example, for the kde/kig [https://edu.kde.org/kig/] project, they use a "critical line" approach: a, find a critical line (means eliminate quadratic terms from the two quadratic equations to get a linear equation), and solve the subsequent linear-quadratic systems. 

2, for all tangential methods, need to be consistent: when tangential methods says "tangential", it should not find two intersection points for the same two entities.

Lots of work needed for each of the items.

r-a-v-a-s, 2015-05-02:
Hopefully it's a lot of *rewarding* work... ;-] Thanks for looking into this.

atorrey, 2015-05-02:
As a thought - If the current approach is finding two possible points, I'd think that the 'correct' point would be on a line between the two points that are found.  Assuming that the error in calculating the actual value was off equally for both found points, the correct point would presumably be halfway in between - so wouldn't a fairly simple approach be to find the two points under the current approach and then calculate the point that would be half way between them?

ART

------------------
Arthur Torrey - <arthur_torrey@comcast.net>
-------------------

https://github.com/LibreCAD/LibreCAD/issues/523#issuecomment-98383414

r-a-v-a-s, 2015-05-02:
I had the same thought. Although that is only a work around. I think you will agree that a proper algorithm will only find one point, since by definition there is only one point of tangency to a given line or circle.

dxli, 2015-05-02:
It's not a problem of detecting intersections.

the problem is that inscribed circle is generated correct up to relative precision of 1e-8, while the tangential points are test at a precision level of 1e-10. Therefore, the inscribed circle actually intersects with one line.

The reasonable way is to improve the equation solver to get the inscribed circle more accurate.

atorrey, 2015-05-02:
I agree it's a workaround, but I'm not sure there IS a better solution.  If I understand the description of the problem that was given earlier, the problem is essentially an 'overflow' of the floating point arithmetic precision where the program is trying to determine the exact location of the points that make up the circle.  I'm not a mathematician, but suspect that it is the same sort of problem one has in determining the 'exact' circumference of the circle, or any other equation that involves pi (or other irrational numbers) 

As such any algorithm is only going to give an approximate answer so the question is how to find something that is 'good enough'...  

ART

------------------
Arthur Torrey - <arthur_torrey@comcast.net>
-------------------

https://github.com/LibreCAD/LibreCAD/issues/523#issuecomment-98416383

atorrey, 2015-05-02:
You will get no argument from me, as I'm anything but an expert....

However I would say as a user, that I'd rather have a function that worked, even if it gives a slightly 'fuzzy' result as that would almost certainly still be better than the alternative of trying to zoom in and 'eyeball' a solution...

ART

------------------
Arthur Torrey - <arthur_torrey@comcast.net>
-------------------

https://github.com/LibreCAD/LibreCAD/issues/523#issuecomment-98416735

r-a-v-a-s, 2015-05-02:
Dongxu, that makes more sense... I knew it would be one of the two. It seems like you are well on your way.

dxli, 2015-05-10:
the tolerance itself is always there, even if we can solve equations at higher precision:

currently, we can generate the tangential circle correct up to 10^-8, but intersection detection is up to 10^-10, causing a tangential point (by the definition of tangential circle) to be detected as intersection points.

because the actual precision is up to the floating point, there's no clear answer close to precision level.

One clear hint, if user specifies to draw with precision 10^-4(as in Current Drawing Preferences), there's no need to report two points 10^-8 apart.

Lots of work need to implement this idea;

atorrey, 2015-05-10:
My girlfriend Mary-Anne is a senior level software engineer in her day job, and I tried to describe this to her, as best I understand it.  What she suggested is that it it really is a rounding problem, she avoids this sort of problem by moving the decimal point - i.e. if the problem is in calculating dollars, try multiplying by 100 and calculating pennies.  Then the error goes away when you shift back to dollars again.

I don't know if that is something that would work, but it might be worth trying...

ART

------------------
Arthur Torrey - <arthur_torrey@comcast.net>
-------------------

https://github.com/LibreCAD/LibreCAD/issues/523#issuecomment-100564236

r-a-v-a-s, 2015-05-10:
The only alternative I could think of is to have an intersection tool that sets special anchor points that the snap to intersection mode will favor over erroneous points in a certain radius.

I see you have done a lot of work already... I wish you luck :D

r-a-v-a-s, 2015-05-30:
I read an interesting comment: 

>The coolest thing I've ever discovered about Pythagorean's Theorem is an alternate way to calculate it. If you write a program that uses the distance form c = sqrt(a^2 + b^2) you will suffer from the lose of half of your available precision because the square root operation is last. A more accurate calculation is c = a * sqrt(1 + b^2 / a^2). If a is less than b, you should swap them and of course handle the special case of a = 0. 

I [asked][1] about this, and a professor of computer science commented:

>It's probably not the square root that's causing the inaccuracies. In 
many other cases, and probably here also, it's the summing of two 
numbers that have vastly different values that loses precision.  A 
demonstration: 

 >>> big = 100000000.0 
 >>> small = 0.000000001 
 >>> (big+small)-big # Should produce a value =small, but gives an exact 
zero instead. 
0.0 

>The squaring of the two values in x*x+y*y just makes the addition even 
more error prone since the squares make large values even larger and 
small values even smaller. 

Regarding our issue, someone suggested the article ["What Every Computer Scientist Should Know About Floating-Point Arithmetic"][2]. 

:four_leaf_clover: 

[1]:https://groups.google.com/forum/#!topic/comp.lang.python/SQy9XVzhyR4
[2]:http://docs.oracle.com/cd/E19957-01/806-3568/ncg_goldberg.html

r-a-v-a-s, 2015-06-25:
What do you think about limiting the length and angle precision to 0.00000 
in Current Drawing Preferences? It seems to "fix" this.
If we start seeing errors past this point, then the extra "precision" is detrimental.

dxli, 2015-06-25:
Yes, we have angular tolerance in radian, 1e-6

r-a-v-a-s, 2015-06-26:
Given an equilateral triangle with a side defined by points (0, 0) and (100, 0),
and using 0.00000 precision, the tangent point of the incircle is found as 50.00000.
However, if the side is defined by points (0, 0) and (1000, 0)
then two points are found: 499.99998 and 500.00002

So I guess its not really much of a fix... :sob:

atorrey, 2015-06-26:
Depending on what units you are talking about, seems reasonable to me, at least for angles... The idea of CAD as I understand it is to make pictures of things that will eventually get turned into physical objects - so is there a value in trying to draw to a greater precision than you can manufacture?

I'm not a complete expert on machining processes, but these days the best subtractive machines are usually only going to about 0.00000" and most are only going to 0.000".  Arguably one or two more places MIGHT be useful...

For angles, I think the usual degree limit is about 3-4 digits, so 5 should work...

ART
------------------
Arthur Torrey - <arthur_torrey@comcast.net>
-------------------

https://github.com/LibreCAD/LibreCAD/issues/523#issuecomment-115413037

r-a-v-a-s, 2015-06-26:
I meant that (499.99998, 0) isn't the midpoint of (0,0) and (1000, 0).
Therefore, if this point is part of a complex geometric construction,
then the following operations are inaccurate;
and we can't use the program to analyze the geometry reliably.

The obvious workaround is to use smaller objects.

I expose weaknesses in systems Art... ;-]

The only thing I'm building is a book about geometry. :sunglasses:

atorrey, 2015-06-26:
True, but until they figure out how to make an 'infinite-bit float' as a number type, we are stuck with the problem you give at SOME point - the question being at what decimal place?

ART

------------------
Arthur Torrey - <arthur_torrey@comcast.net>
-------------------

https://github.com/LibreCAD/LibreCAD/issues/523#issuecomment-115492794

r-a-v-a-s, 2015-06-27:
Dongxu, could [Boost.Multiprecision][1] help?

http://hvks.com/Numerical/arbitrary_precision.html
https://gmplib.org/
http://www.mpfr.org/
http://mpir.org/index.html

[1]:http://www.boost.org/doc/libs/1_58_0/libs/multiprecision/doc/html/index.html

dxli, 2015-06-27:
currently, double is used in saved files (correct me, if it's not the case).

internally, all entities are double in coordinates, angles, radius, etc.

changing everything to "long double" should be an easy extension for better precision.

To actively use multiple precision would require lots of implementation, if it's doable at all.

r-a-v-a-s, 2015-08-01:
>the problem is that inscribed circle is generated correct up to relative precision of 1e-8, while the tangential points are test at a precision level of 1e-10.

What functions contain the relevant code?

dxli, 2015-08-04:
snap point is by: RS_EntityContainer::getNearestIntersection
the actual code to generate tangent point is in each action class, like RS_ActionDrawTan3

All the equation solvers are in RS_Math:: like, RS_Math::QuadraticSolver(), etc.

The equation solvers need some improvement:

1. review of Cubic and quartic solvers; Could be as simple as what we do for quadratic, try to use $ -b c / ({\sqrt{a^2/4 - bc} + a/2})$ instead of $-a/2+\sqrt{a^2/4-bc}$, for $a>0$

2. Quadratic-Quadratic system solver needs a new algorithm. The current solver convert it to a quartic equation in one step. A better algorithm could be:

a. find lines passing intersections;
b. solve linear-quardatic systems.

3. blindly use "long double" instead of "double" for equation solvers.

r-a-v-a-s, 2015-08-05:
谢谢 黎東旭

[rs_actiondrawcircletan3.cpp][6]
[rs_entitycontainer.cpp][3]
[lc_quadratic.cpp][4]
[rs_circle.cpp][1]
[rs_math.cpp][2]
[rs_arc.cpp][7]

Something I just learned:
- press `t` on the [main page][5] of this repository, or bookmark [File Finder][8]
- type in the name of a file to search for

[1]:https://github.com/LibreCAD/LibreCAD/blob/master/librecad/src/lib/engine/rs_circle.cpp
[2]:https://github.com/LibreCAD/LibreCAD/blob/master/librecad/src/lib/math/rs_math.cpp
[3]:https://github.com/LibreCAD/LibreCAD/blob/master/librecad/src/lib/engine/rs_entitycontainer.cpp
[4]:https://github.com/LibreCAD/LibreCAD/blob/master/librecad/src/lib/math/lc_quadratic.cpp
[5]:https://github.com/LibreCAD/LibreCAD
[6]:https://github.com/LibreCAD/LibreCAD/blob/master/librecad/src/actions/rs_actiondrawcircletan3.cpp
[7]:https://github.com/LibreCAD/LibreCAD/blob/master/librecad/src/lib/engine/rs_arc.cpp
[8]:https://github.com/LibreCAD/LibreCAD/find/master

dxli, 2015-08-05:
There's a file name filter in the bottom left corner of Qt Creator, by default.

r-a-v-a-s, 2015-08-05:
Yes more info: http://doc.qt.io/qtcreator/creator-editor-locator.html

I'm just exposing the information to people that might want to help by reviewing the code.

r-a-v-a-s, 2015-10-08:
What about `RS_Information::getIntersectionLineArc`?
It seems like it is unused.

dxli, 2016-01-16:
started to test the degenerate quadratic form idea:
https://github.com/dxli/LibreCAD/commits/kig

2d298181bb1e

So far, the code ported from KDE/kig works great in finding a degenerate quadratic form, but failed to correctly locate the radical lines from the degenerate quadratic form:

if M is a  3x3 symmetric matrix of quadratic form ( (x, y, z) M (x, y, z)' ), and det|M| = 0, the quadratic form can be reduced to a product of two linear forms.

The ported kig code fails to reduce a simple form:
1 0 0
0 -1 0
0 0 0

which is x^2 - y^2  = (x + y) (x -y)

I will try to rewrite this part to see whether we can improve robustness of the LibreCAD quadratic-quadratic solver.

r-a-v-a-s, 2016-01-16:
I appreciate your effort! :sunglasses: :+1: 
[祝你好运。](https://tatoeba.org/eng/sentences/search?query=good+luck&from=und&to=und)

dxli, 2016-01-24:
After a review of the kig algorithm and implementation, it seems:

* their implementation is not compact, and difficult to maintain, but well commented at least;
* their handling of degenerate quadratic form is a little too complex: too much manual operations,*i.e.* if/else, for the simplicity of the problem itself;
* more on the previous point, we can rely on a robust eigen system to decompose a reducible quadratic  form into product of two linear forms here;

looks like we have to rewrite every part of the algorithm:

* find combination to form a degenerate quadratic form, using our own cubic equation solver;
* given a degenerate quadratic form, find Eigen decomposition of the matrix. The trouble, we need a library to do it, or have to implement one for 3x3 symmetric matrices(and probably, only for rank 2). We have to decide whether we pull in a library(or just a function), or  to do it manually;
* after eigen decomposition, the linear form is trivial to recover:

x^2 - y^2 = (x+y) (x-y)

r-a-v-a-s, 2016-01-24:
Is there anything we can use from boost.geometry?
http://www.boost.org/doc/libs/1_60_0/libs/geometry/doc/html/geometry/matrix.html

I noticed r-tree under spatial indexes.

dxli, 2016-01-24:
yes, rtree should be great for intersections. Compare:

* brute force, try all pairwise intersections for N entities, ~ N^2/2
* rtree, only handle intersections for pairs with overlapped bounding boxes. Suppose the average case of uniformly distributed bounding boxes, it takes about  lg_(M/2) (N) steps to locate all intersected bounding boxes. The actual number of pairs to calculate is about N*ratio, with the ratio as the average bounding box area over the area of the union of all boxes.. This is great for splinepoints. Say, there are huge number of segments(entities), but all bounding boxes are all small.
* for entities with large bounding boxes, rtree doesn't help, consider diagonal straight lines, the bounding box covers the whole view

r-a-v-a-s, 2016-01-24:
Something else possibly worth investigation is the "Eigen" library.
http://forum.librecad.org/geometry-libraries-td5711753.html

vlachoudis, 2016-01-25:
@r-a-v-a-s in my program flair [http://www.fluka.org/flair] the core of the 2D viewer/debugger is based on a conic-conic intersection. I've spend a lot of time in checking the numerical robustness of the algorithm. If you want you can give a look on the "geoviewer" add-on module especially the conic.cc Conic::intersect() and bmath.cc quartic equations. (You can download the code from the svn repository, see the download section for information).

I am not using a fixed precision for the problem, but most of the critical operations (e.g. quartic equation solving) calculate both the result and an upper estimate of the error from the resulting operation, for the subsequent decision operations.

The loss of precision of cubic and quartic equations is dramatic. No matter what you will do the cubic will result 15/3 useful digits and the quartic 15/4 useful digits for a double precision number. For this reason all solutions in the end, both for the cubic and quartic operations,  are refined by an iterative Newton approach.

For the conic-conic intersection I've tried two algorithms, one is direct substitution of one equation in the other and the second was to create a pencil of conics that reduces the final equation to a cubic one.

r-a-v-a-s, 2016-01-25:
Thanks a lot. Although, dxli is the one who understands this topic.
I mostly work on the user interface. ;-]
:v:

dxli, 2016-01-25:
hi @vlachoudis 

Thanks a lot, and I had quick look at the source code, and the conic-conic intersection algorithm is by direct substitution, and with careful handling of robustness issues.

Just one minor but interesting point on the underneath quadratic solver, do we need to calculate sqrt(a^2 + b^2) as a*sqrt(1 + (b/2)^2), given abs(a)>=b and abs(a) >0. Someone mentioned this before, and I think it might be a good point for your code as well.

I am interested in the pencil of conics method because the substitution method seems to be a great cause of precision loss, and using cubic is hopefully better than quartic. After cubic equation solving, steps are quite robust: find the linear forms, and find line-conic intersections.

vlachoudis, 2016-01-26:
@dxli I've implemented both. The one that is enabled now is the direct substitution. The pencil code is disabled under an `#if 0`.

To be honest I've got worst performance with the pencil of conics (but this was before I implemented the newton refining of the cubic/quartic) solutions and maybe because I didn't expand and optimize all matrix and conic operations. I haven't test it afterwards.

In the end both solutions have the same loss of precision, since the quartic to be solved, you need to  eliminate the cubic term, resulting to a cubic equation and a quadratic.

With the pencil of conics you create the pencil C1+l*C2=0 which is a conic that intersects the initial conics on the points of intersection. You solve with a cubic equation for l  (determinant D=0) to degenerate it to a line-conic, and then you have to do line-conic intersections (for all l real solutions) with any of the two initial conics, needing a quadratic.

I will give a try the sqrt(a^2+b^2). But I am afraid that the big loss of precision comes from the multiple sqrt() required. Each one halves the number of useful digits. Unless if you could use "long double" sqrtl() with a performance penalty.

dxli, 2016-01-26:
@vlachoudis 

Sorry for missing the pencil part.

I had a quick review of Conic::splitLines(), and I feel the robustness of the splitting algorithm can be improved using QR type algorithms.

I don't have solid proof, but my feeling is drawn from the type of difference between Gram-Schemidlt and Householder procedures. 

If this reasoning is good, eigen vector based algorithm can simplify the algorithm of splitting while being numerically stable.

in short, in homogeneous coordinates, the diagonalized M is trivial to split, and apply back to the original frame to find the linear forms.

vlachoudis, 2016-01-26:
@dxli I didn't try to calculate the eigen vectors, since it would require the solution of cubic equation or iterative QR.
For the conics it is simpler to normalize them, I have the function "theta()" that calculates the angle to rotate the conic in order to eliminate the "h" component and with a translation you can bring it to the origin removing the "g" and "f" components. Actually I am doing this on the "set()" method when I calculate the invariants of the conics. Once normalized splitting into two lines is easy.

In my approach I tried to avoid all the trig functions for normalizing the conic. But it is true I never test it if it would perform better. Maybe I can give it a try.

vlachoudis, 2016-01-28:
@dxli I've implemented a new KahanSum class for improving the precision loss in multiple additions. I used it to calculate the coefficients of the conic-conic intersection and it did improve with a bit of CPU penalty. In one of my complicated test cases, km of LHC geometry with 10xmillions of conic-intersections it reduced the problematic cases to half.

dxli, 2016-01-28:
@vlachoudis 

With understanding from this discussion, I should go ahead to implement the conic pencil algorithm and some test cases for comparison with the substitution/quartic method.

Looks like Kahan Summation should be used. I'm also thinking about using C++ 128bit "long double" type for intermediate steps.

one more question, if we want to integrate your code, is there any licensing issue. From what I read, the license terms cover your code are not compatible with GPLv2+.

vlachoudis, 2016-01-28:
@dxli of course you can use part of my code. Can you keep my copyright notice on the top and say that I've granted you the permission to use it.
The conic pencil algorithm its long time that I've wrote it and would require some optimization.
Kahan summation helps, and if you use "long double" it would be even better with a some CPU penalty, if I understand correct the "long double" operations are done by software.

vlachoudis, 2016-01-28:
@dxli I've just tested it again the pencil of conics, it works but it failed in some nasty cases. As I said I've spend more time polishing the substitution and dealing with precision problems, I believe it can improved.

dxli, 2016-01-29:
@vlachoudis 

Following the discussion, I did a clean up for our quadratic solver:

* sqrt(a^2 \pm b^2) trick;
* void loss of significance;
* long double type intermediate

82ea78ecd3f5a0

vlachoudis, 2016-01-30:
@dxli the whole discussion made me re-look again all precision problems I have in my program.

To help me spot the precision lost I've made yesterday a very small template class that creates two types `Float` and `Double` that behave as normal `float` and `double`, but at the same time they duplicate all operations with a `long double` variable. From the difference you can estimate the precision lost at every moment of the operation. Is quite useful for debugging precision errors.

You can find it at [https://github.com/vlachoudis/fperror]

r-a-v-a-s, 2016-01-30:
Based on the latest code:
Tangent points found using an equilateral triangle with the origin (0,0) as the left base point,
and the triangle's incircle:
[image]

side length = 10
- 5.00000, -0.0000 (why negative?)
- 7.50000, 4.33013
- not found

side length = 100
- not found
- not found
- not found

side length = 1000
- not found
- 750.000, 433.013
- not found

dxli, 2016-01-30:
[image]

@r-a-v-a-s 
It actually works well here.

I feel there's a subtle bug in the experimental quadratic solver. I will review it and add more test cases.

dxli, 2016-01-31:
@vlachoudis 

I started to put together an implementation, hoping to be clear and compact. Need more time to work on unit tests and documentation.

I added Kahan summation: 475c238716f4e

For the pencil part, the implementation seems to be compact on eigen decomposition and linear forms. fec2731278db5

Not finished yet, and we will see.

vlachoudis, 2016-02-01:
@dxli Great. I don't know if you saw, but in the last svn I've created a KahanSum class that you don't need to give a list as an argument. I believe it is simpler to use. You could use it also for the pencil of conics for calculating the linear combination coefficients.

r-a-v-a-s, 2016-02-11:
A warning worth noting:
```
rs_math.cpp:432: warning: absolute value function 'fabs' given an argument of type 
'const long double' but has parameter of type 'double' which may cause truncation of value 
[-Wabsolute-value]
        long double const fc = fabs(c);
                               ^
```

```
rs_math.cpp:432: use function 'std::abs' instead
        long double const fc = fabs(c);
                               ^~~~
                               std::abs
```
The same for line 452.

r-a-v-a-s, 2016-02-15:
Same test on iMac / OS X:

side length = 10
- not found
- 7.500000, 4.330127
- 2.500000, 4.330127

side length = 100
- 50.00000, -0.00000 (negative 0... maybe it lives in counterspace? ;D)
- 75.00000, 43.30127
- 25.00000, 43.30127

side length = 1000
- not found
- 750.000, 433.013
- 250.000, 433.013

More points found overall, and an extra digit for the 10 and 100 cases.

dxli, 2016-02-16:
@vlachoudis 

my eigen space approach works in finding all solutions, but the precision is still bad.

https://github.com/dxli/LibreCAD/tree/kig

The implementation is quite compact to me, (ignoring the debugging portion).

Maybe, I should tweak the cubic solver for precision first.

vlachoudis, 2016-02-16:
@dxli cubic and quartics are a killer on precision. Out of a cubic equation you will get roughly 5-6 good digits (it comes from the cubic root). For this reason If you see on my implementation I am refining the solutions of the cubic by a repetitive Newton approach. After 3-4 iterations if it converges it will give you ~10 good digits of precision.
Also in your linearReduction you are solving for the eigenvalues to normalize the matrix of the conic. In this step you can lose a lot of precision. I would recommend either
i) calculate all possible combinations you have to solve in the worst case 2 quadratic equations, like what I am doing in my code.
ii) if you still want to normalize, first rotate the matrix with an angle theta=0.5*atan2(2*h,a-b)
  and then translate by -g,-h. Look in my code the conic.cc especially the :set(), which sets the variables, calculates the invariants of the conic, and calculates the new coefficients after the rotation and translation.

dxli, 2016-02-16:
some trivial testcases, I didn't try to send in hard cases.

I took to line equations, form the quadratic for doing outer product(M), and compare the linear reduction from the symmetric (M + transpose(M)) quadratic form, find the maximum difference (norm_inf) in linear forms from the input
'''
1*x + 2*y + 1 = 0 2*x + 3*y + 2 = 0 err = 2.21997e-10
1*x + 2*y + 1 = 0 1*x + 4*y + 5 = 0 err = 6.78831e-11
1*x + 2*y + 1 = 0 4*x + 1*y + 7 = 0 err = 1.37882e-09
1*x + 2*y + 1 = 0 3*x + 1*y + 3 = 0 err = 4.19599e-10
1*x + 2*y + 1 = 0 1*x + 1*y + 4 = 0 err = 1.79403e-09
1*x + 2*y + 1 = 0 1*x + 10*y + 1 = 0 err = 2.24328e-10
1*x + 2*y + 1 = 0 10*x + 1*y + 11 = 0 err = 2.34026e-10
1*x + 2*y + 1 = 0 4*x + 0*y + 7 = 0 err = 8.03202e-10
2*x + 3*y + 2 = 0 1*x + 4*y + 5 = 0 err = 1.0925e-08
2*x + 3*y + 2 = 0 4*x + 1*y + 7 = 0 err = 3.72449e-10
2*x + 3*y + 2 = 0 3*x + 1*y + 3 = 0 err = 1.26722e-10
2*x + 3*y + 2 = 0 1*x + 1*y + 4 = 0 err = 9.81391e-10
2*x + 3*y + 2 = 0 1*x + 10*y + 1 = 0 err = 7.43923e-10
2*x + 3*y + 2 = 0 10*x + 1*y + 11 = 0 err = 8.93543e-09
2*x + 3*y + 2 = 0 4*x + 0*y + 7 = 0 err = 2.14982e-09
1*x + 4*y + 5 = 0 4*x + 1*y + 7 = 0 err = 9.933e-10
1*x + 4*y + 5 = 0 3*x + 1*y + 3 = 0 err = 6.5386e-09
1*x + 4*y + 5 = 0 1*x + 1*y + 4 = 0 err = 1.18852e-10
1*x + 4*y + 5 = 0 1*x + 10*y + 1 = 0 err = 1.10509e-08
1*x + 4*y + 5 = 0 10*x + 1*y + 11 = 0 err = 9.3393e-09
1*x + 4*y + 5 = 0 4*x + 0*y + 7 = 0 err = 2.39544e-09
4*x + 1*y + 7 = 0 3*x + 1*y + 3 = 0 err = 2.37669e-10
4*x + 1*y + 7 = 0 1*x + 1*y + 4 = 0 err = 2.35021e-10
4*x + 1*y + 7 = 0 1*x + 10*y + 1 = 0 err = 1.51859e-09
4*x + 1*y + 7 = 0 10*x + 1*y + 11 = 0 err = 8.02098e-10
4*x + 1*y + 7 = 0 4*x + 0*y + 7 = 0 err = 1.71759e-10
3*x + 1*y + 3 = 0 1*x + 1*y + 4 = 0 err = 3.50922e-10
3*x + 1*y + 3 = 0 1*x + 10*y + 1 = 0 err = 5.20855e-10
3*x + 1*y + 3 = 0 10*x + 1*y + 11 = 0 err = 5.90035e-11
3*x + 1*y + 3 = 0 4*x + 0*y + 7 = 0 err = 3.11422e-09
1*x + 1*y + 4 = 0 1*x + 10*y + 1 = 0 err = 2.0853e-09
1*x + 1*y + 4 = 0 10*x + 1*y + 11 = 0 err = 1.55702e-10
1*x + 1*y + 4 = 0 4*x + 0*y + 7 = 0 err = 8.56697e-11
1*x + 10*y + 1 = 0 10*x + 1*y + 11 = 0 err = 1.79067e-09
1*x + 10*y + 1 = 0 4*x + 0*y + 7 = 0 err = 1.29262e-09
10*x + 1*y + 11 = 0 4*x + 0*y + 7 = 0 err = 7.78921e-12
'''

vlachoudis, 2016-02-16:
@dxli I did a similar test with your equations created the quadratic/conic form, split to two lines and multiply them to get the combined coefficients and normalize them to the highest value. I've tested the diff between the coefficients and I got the following maxerr

1 2 1 2 3 2 maxerr=0
1 2 1 1 4 5 maxerr=0
1 2 1 4 1 7 maxerr=0
1 2 1 3 1 3 maxerr=0
1 2 1 1 1 4 maxerr=0
1 2 1 1 10 1 maxerr=0
1 2 1 10 1 11 maxerr=0
1 2 1 4 0 7 maxerr=0
2 3 2 1 4 5 maxerr=0
2 3 2 4 1 7 maxerr=0
2 3 2 3 1 3 maxerr=0
2 3 2 1 1 4 maxerr=0
2 3 2 1 10 1 maxerr=0
2 3 2 10 1 11 maxerr=0
2 3 2 4 0 7 maxerr=0
1 4 5 4 1 7 maxerr=0
1 4 5 3 1 3 maxerr=1.776356839400250464678e-15
1 4 5 1 1 4 maxerr=0
1 4 5 1 10 1 maxerr=0
1 4 5 10 1 11 maxerr=7.105427357601001858711e-15
1 4 5 4 0 7 maxerr=0
4 1 7 3 1 3 maxerr=0
4 1 7 1 1 4 maxerr=0
4 1 7 1 10 1 maxerr=0
4 1 7 10 1 11 maxerr=0
4 1 7 4 0 7 maxerr=0
3 1 3 1 1 4 maxerr=0
3 1 3 1 10 1 maxerr=0
3 1 3 10 1 11 maxerr=0
3 1 3 4 0 7 maxerr=0
1 1 4 1 10 1 maxerr=0
1 1 4 10 1 11 maxerr=7.105427357601001858711e-15
1 1 4 4 0 7 maxerr=0
1 10 1 10 1 11 maxerr=1.421085471520200371742e-14
1 10 1 4 0 7 maxerr=0
10 1 11 4 0 7 maxerr=0

dxli, 2016-02-16:
@vlachoudis 

conclusive :)

I initially didn't like your direct reduction, because it doesn't treat all terms equally, for example, the term F is not used, and assumed to be  2*F = f1*c2 + f2*c1.

The results are conclusive enough. I will port your code directly from here.

[6 more replies]