LibreCAD/LibreCAD #528: layer activation added to undo stack

r-a-v-a-s, 2015-05-15:
The recent change to the undo system regarding
*Application Preferences → Defaults → Modify layer of selected entities, at layer activation*
makes it so every click on a layer is added to the undo stack.

kromanenko, 2015-05-15:
Please clarify actions by more detailed way, i could not reproduce. 
BTW, creation of layer can not be reverted, that`s looks like a bug.

kromanenko, 2015-05-15:
On new commit 9f6f40070a227b6c7f79cea66b8501206c7ba004 it is also reproduced.

r-a-v-a-s, 2015-05-15:
- create a circle
- click on a layer three times

You will have to undo 4 times for the circle to be removed.

r-a-v-a-s, 2015-05-21:
This bug still exists; and it adds a lot of pollution to the undo stack for anyone who uses layers.
Note that disabling "Modify layer of selected entities, at layer activation" fixes the issue.

I'm still an advocate for replacing "Modify layer of selected entities, at layer activation" with:

1. right click context menu for layers has "move selected here" 

2. make a hotkey, command and/or top level icon for Modify → Attributes

atorrey, 2015-05-21:
Sort of related, though probably not directly...

One of my minor annoyances, and it isn't a show stopper, but just slows me down, is that if I want to modify something, I tend to select the entities first, then select the modification.  It is annoying that the modification step then brings up the select menu that I have to click through....

Would it make any sense to have the modification steps check to see if entities have been selected, and if so, skip the select menu step, and if not then bring up the menu?  That way it is efficient whichever way you do your workflow...

ART

------------------
Arthur Torrey - <arthur_torrey@comcast.net>
-------------------

https://github.com/LibreCAD/LibreCAD/issues/528#issuecomment-104391610

dxli, 2015-05-22:
atorrey,

have a look at commit: 872e3e52a
Please help to improve this feature.

dxli, 2015-05-22:
In general, the undo cycle needs some enhancement.

atorrey, 2015-05-22:
I'm not a code person, but if I'm reading it right, looks like you have a good solution...

I need to figure out how to download the version with this in it to test, but not tonight...

Many thanks for working on this!

ART

------------------
Arthur Torrey - <arthur_torrey@comcast.net>
-------------------

https://github.com/LibreCAD/LibreCAD/issues/528#issuecomment-104480178

dxli, 2015-05-22:
ART,

for windows/mac, nightly builds can be found at: https://sourceforge.net/projects/librecad/files/
In general, you can follow our wiki to build from latest source:
http://wiki.librecad.org/index.php/LibreCAD_Installation_from_Source

lordofbikes, 2018-10-28:
With latest fixes to undo/redo code, 2d8e025 to ed90a19, it should be more stable now.
But this doesn't change the primary issue.
Also there is nothing wrong with that behavior in my eyes. After selecting a layer, with entities selected, the entities are moved to the selected layer and the selection is revoked. Only when entities are selected again the click on the layer list adds a new undo cycle. Well, this may be different before.
But now, if one repeats this multiple times, I see nothing wrong when each cycle is added to the undo stack.

closed 2018-10-28 by lordofbikes