LibreCAD/LibreCAD #530: Layer creation can not be reverted

kromanenko, 2015-05-15:
Steps:
1. Open new drawing.
2. Create new layer.
3. Note that Undo button and menu item are still disabled. Expected to be able to undo layer creation.

Env: Ubuntu 14.04, build on commit 140eaee3bd0610f49a848e1cce539d0ab008660d

kromanenko, 2015-05-15:
On new commit 9f6f400 it is also reproduced.

dxli, 2015-05-21:
Currently, "Undo/Redo" are for entities only, while Layer is a property of entities, but not an entity *per si*.
Therefore, this feature can not be easily supported.

r-a-v-a-s, 2015-08-10:
It doesn't seem worth the added complexity.
I suggest keeping the undo stack as simple / clean as possible.

lordofbikes, 2018-10-28:
After getting my teeth into undo code I come to the same conclusion.
Undo/redo stack is implemented in entities, layer creation/removal is something completely different.
So to revert layer creation, don't use undo, but remove layer.
Concerning redo layer removal see #531

closed 2018-10-28 by lordofbikes