LibreCAD/LibreCAD #532: segmentation fault

r-a-v-a-s, 2015-05-22:
The first crash happened when I was deleting an overlapping circle after using modify-rotate to create multiple copies of a circle. The second crash was right after pressing OK while trying to rotate/create multiple copies of a circle.

[image]
[image]

I'm sensing this has something to do with line 64... ;-]

dxli, 2015-05-22:
detailed steps to reproduce this bug?

r-a-v-a-s, 2015-07-03:
Perhaps your change to the debugging on line 64 fixed it. I have not been able to reproduce this.

closed 2015-07-03 by r-a-v-a-s

dxli, 2015-07-03:
Cad toolbars, always ugly.