LibreCAD/LibreCAD #534: Feature Request: GUIless converter similar to qcad2's vec2web

tumagonx, 2015-06-03:
In master branch there is makercamSVG generator, is there  any plan to make such converter. It would be very handy since libdxfrw is the best DXF/DWG library around.

input: DXF, DWG, SHP
ouput: SVG, PS/PDF, Bitmap

dxli, 2015-06-03:
this can be implemented in v3 also

dxli, 2015-06-30:
Added this one to my pipeline

Rallaz, 2015-06-30:
A simple GUIless converter with
input: DXF, DWG
ouput: DXF
are writen as example in libdxfrw
http://sourceforge.net/p/libdxfrw/code/ci/master/tree/dwg2dxf/
May be adding other output ?

dxli, 2015-06-30:
Hi Rallaz,

It's still better, if the feature can be provided by LibreCAD, from command line.

We can convert dwg to dxf, svg(xml);

also, can convert dwg/dxf, to PDF, png, etc.

dxli

tumagonx, 2015-06-30:
also to do complex text/hatch rendering via librecad first

Antinatalist, 2018-02-16:
is there anything on github like:

input: SVG, PS/PDF, Bitmap
output: DXF, DWG, SHP