LibreCAD/LibreCAD #543: Small toolbar icon size on HIDPI screens

pacharanero, 2015-07-12:
On HIDPI/Retina screens (such as on my Dell Precision M3800 which has a 3200 x 1800 display) the toolbar icons are rendered very small so that it causes eyestrain to see them. I don't know if this is something that is caused by upstream Qt4/5 issues with HIDPI displays or whether this can be dealt with in LibreCAD with a 'toolbar icon size' setting.

dxli, 2015-07-15:
hi pacharanero,

we are aware of this problem, even though none of the main developers have a HD monitor to do the test, so we will need your help here.

1, please confirm the extent of this problem, we understand the problem exists in "cad toolbars" in the left, layer/block widgets in the right;

2, do we have other issues on HD, such as font size?

Thanks!

pacharanero, 2015-07-16:
Hi dxli

this affects all icons/buttons but fonts are the right size

here's a screenshot of the basic full-screen LibreCAD window as it appears when I open the application on my Dell M3800. If you set up the zoom factor when you view it so that the window takes up the full screen (excluding Cinnamon/Gnome panel) then you will see what I see

[image]

pacharanero, 2015-07-16:
interestingly QCAD (which I understand is the package that LibreCAD is forked from) has an option in the settings to allow larger icons.

r-a-v-a-s, 2015-07-16:
I have already implemented this option on my own fork. However, it does not use the CAD toolbar. I'm not sure how it's going to work on all systems with the CAD toolbar considering → [this][1].
I did add a checkbox to allow for the system default. I can add it here if the other developers want that.

[1]:http://forum.librecad.org/Not-working-well-on-Mac-OSX-anymore-td5711800.html#a5711802

dxli, 2015-07-17:
ravas:

could there be a simple resolution: 

http://blog.davidedmundson.co.uk/blog/kde_apps_high_dpi

r-a-v-a-s, 2015-07-17:
:point_right: [more info][1]

[1]:http://stackoverflow.com/questions/24367355/automatic-rescaling-of-an-application-on-high-dpi-windows-platform

r-a-v-a-s, 2015-10-01:
@pacharanero What operating system are you using? The icon size option is available in the master branch. I'm curious what size works for you.

pacharanero, 2015-10-01:
I'm on Linux Mint and I think I installed LibreCAD from the standard Ubuntu
repos. Maybe it is behind the master branch?

I'll try to find out what version.

M

pacharanero, 2015-10-01:
Linux Mint 17.2 is based on Ubuntu 14,04 LTS so is very likely to have had
an older version of LibreCAD in its repos.

I've installed the PPA and will install the latest stable and go from
there. Thanks for the pointer to the version being the likely problem.

M

pacharanero, 2015-10-01:
I installed from stable PPA and have version 2.0.8 of libreCAD.
I can't find an icon size setting tho'

M

dxli, 2015-10-01:
this feature is on the master branch, but not 2.0

r-a-v-a-s, 2015-10-01:
http://librecad.org/cms/home/installation/linux.html
`$ sudo add-apt-repository ppa:librecad-dev/librecad-daily`

Does this use the master branch?

r-a-v-a-s, 2016-09-07:
The ppa seems to have been abandoned,
and the owner seems to ignore my questions.
Although the daily ppa does have an older 2.2.0 alpha build.

2.1.0 was released on June 4 (and later 2.1.1);
and the latest 'official' package I can find for debian (and forks) is 2.0.9...
I guess no one is interested enough.

You could build from source:
https://github.com/LibreCAD/LibreCAD/wiki/Build-from-source

closed 2016-09-07 by r-a-v-a-s