LibreCAD/LibreCAD #546: Feature Request: Zoom in and out on cursor using + and - keys

membiblio, 2015-07-14:
Other cad packages allow you to zoom in and out on the current cursor location using the + and - keys which is very helpful when using the mouse to draw precise objects as you can zoom in on the mouse cursor to place objects precisely.

r-a-v-a-s, 2015-07-14:
Just to make sure you are aware of all the current options:

* scrolling the mouse wheel
* ctrl & - / ctrl & +
* view → window zoom

dxli, 2015-07-15:
let me talk with one example: 

0, you have a single circle (0,0) r=1, and by using "zoom auto", the view would be centered at (0,0)
1,  start to zoom in, in several steps, the circle curve would be out of view
2, the new feature: if mouse cursor is on (1,0), keep zooming in, you can always see the curve at roughly the same position in view, and slowly turns into a vertical line

dxli, 2015-08-04:
I will make Ctl (+/-) to zoom around the current mouse pointer (if the cursor is within the drawing area)

r-a-v-a-s, 2015-08-10:
Is this Ctrl (+/-) feature acceptable for you?
We might be able to work around it,
but currently all character producing keystrokes are sent to the command line.

membiblio, 2015-08-12:
Ravas,

     I'm guessing Ctrl +/- might be ok - it is a place to start - though 
other cad packages just use +/- to zoom in and out around the cursor.  
So my wish is that ctrl +/- zooms and keystrokes go to the interpreter 
when that command window is in focus and maybe a key to focus on that 
command window. Your thoughts?

Richard

r-a-v-a-s, 2015-08-12:
My first thought is to make two modes.
The first mode sends keystrokes directly to the command line (like now).
The second mode allows for a commandline focus hotkey,
and single keystrokes could be tool hotkeys.

I would bet that simply replacing mode 1 with mode 2 won't happen.

membiblio, 2015-08-12:
Ravas,

     I really like your thinking here - simply spectacular idea. If you 
can do what you are describing - could you please do that?  I believe it 
will change this product for the absolute better - a new and better 
direction for the mmi. JMHThoughts :)

Richard

severoraz, 2015-09-07:
It appears to me that this issue is solved and should be closed, is it not?

membiblio, 2015-09-07:
Severoraz,

     What was your final fix?   If done I will dl new git today, 
compile, install, test and say "Thank You!" :)

Richard Newman
Pittsburgh PA USA

severoraz, 2015-09-07:
Well I have not participated in the solution of this issue, but I use LibreCAD 2.0.8 and when and I zoom in or out using <Control>+ or <Control>- the cursor position in relation with the viewport remains constant

membiblio, 2015-09-10:
> LibreCAD 2.0.8 and when and I zoom in or out using + or - the cursor

Severoraz,

     Regarding the +/- to zoom in and out - it works perfectly and it 
works as you expect it to. I would say that it is absolutely perfect!   
This, in my opinion, will be a major feature as now LibreCad works as 
other commercial and, especially, open source cad type solutions that 
are similar, but different, like KiCAD.

     Big Big Kudos to the developer!  I give it a A+++++ :)

Richard Newman
Pittsburgh PA USA

closed 2016-03-15 by r-a-v-a-s