LibreCAD/LibreCAD #548: Tangential 3 Circles problem case

r-a-v-a-s, 2015-07-17:
The circle in the upper half of the [vesica piscis][1] is not considered for "Tangential 3 Circles".

[image]

[1]:https://en.wikipedia.org/wiki/Vesica_piscis

r-a-v-a-s, 2015-07-17:
Actually it is considered if the two circles have a radius of 100, but not when their radius is 10.

dxli, 2015-07-17:
can you send in the dxf file?

I can get 6 possible tangential circles here.

dxli, 2015-07-17:
I can still get all 6 possible solutions with your dxf file

r-a-v-a-s, 2015-07-17:
:astonished:

dxli, 2015-07-27:
Looks like we can expand this tangent3 feature to include "point" entities to provide features like:

draw a circle passing 3 given points;
draw a circle passing 2 points and tangential to a given line/circle
draw a circle passing 1 point and tangential to two given lines/circles

dxli, 2015-07-28:
allow point entities for "tan3": 73cab7e

r-a-v-a-s, 2015-07-29:
Now I get all solutions. :+1:

closed 2015-07-29 by r-a-v-a-s