LibreCAD/LibreCAD #549: crosshair not shown on tool activation

r-a-v-a-s, 2015-07-21:
If I create a line and then enable 'snap to middle' 
and select the vertical or horizontal line tool, 
there is no cross-hair to show the snapping. 
I can click on the line and it will produce what I want, 
and then the cross-hair will appear.

`RS_ActionDrawLineAngle`
* ActionDrawLineAngle
* ActionDrawLineHorizontal
* ActionDrawLineVertical

The only relevant thing I noticed different from `RS_ActionDrawLine` was the missing `drawSnapper();` in the `init` function. However, adding it didn't fix the issue. 

Other line tools that do not show a crosshair on activation:
* parallel
* parallel through
* bisector
* tangent (c,c)
* orthogonal
* relative
* freehand

with circles:
* concentric
* all tangential tools

An observation: most of these tools use the option toolbar and/or do not ask for a point as the first step.

dxli, 2015-07-27:
I will have fix this one.

dxli, 2015-07-27:
fixed freehand by commit: 45900bba52

dxli, 2015-07-27:
crosshair is used to show for a snapping point, but not for entity catching.

atorrey, 2015-07-28:
Another minor but helpful feature would be if there was some sort of automatic change in the crosshair color when it is overlapping an existing feature, and / or if placing a horizontal / vertical line to be different from the line being placed.

I find that if the crosshair is the same color as other drawing features it can be harder to spot where the line you are trying to add is going - especially if the scale is such that the legs of the crosshair are longer than the line being placed...

Perhaps some sort of calculation that if the crosshair is intersecting a drawing feature it should have some addition to it's color value, and if it is placing a horizontal / vertical line that it be something other than the line color...

ART

------------------
Arthur Torrey - <arthur_torrey@comcast.net>
-------------------

https://github.com/LibreCAD/LibreCAD/issues/549#issuecomment-125374674

dxli, 2015-07-28:
ART,

I will see whether there's a good solution for that.

r-a-v-a-s, 2015-07-29:
Art, while your idea is related, it's still off topic and will be somewhat hidden after I close this. 
I suggest opening a separate issue, or posting in the suggestions & feedback section on the forum.

r-a-v-a-s, 2015-07-29:
Thanks dxli. :+1:

closed 2015-07-29 by r-a-v-a-s

atorrey, 2015-07-29:
OK

------------------
Arthur Torrey - <arthur_torrey@comcast.net>
-------------------

https://github.com/LibreCAD/LibreCAD/issues/549#issuecomment-126059056