LibreCAD/LibreCAD #550: Make crosshair change color to contrast w/ other drawing features?

atorrey, 2015-07-29:
Another minor but helpful feature would be if there was some sort of automatic change in the crosshair color when it is overlapping an existing feature, and / or if placing a horizontal / vertical line to be different from the line being placed. 

I find that if the crosshair is the same color as other drawing features it can be harder to spot where the line you are trying to add is going - especially if the scale is such that the legs of the crosshair are longer than the line being placed... Perhaps some sort of calculation that if the crosshair is intersecting a drawing feature it should have some addition to it's color value, and if it is placing a horizontal / vertical line that it be something other than the line color... 

I had commented on this as a part of another problem, and was requested to open as a separate item, so have copied the initial comment.

ART

r-a-v-a-s, 2015-07-29:
I like the idea;
although, I'm wondering if the added amount of analysis would be worth the convenience.

What about a hotkey to toggle the appearance of the crosshair?
Changing the color would be the most simplistic;
or we could design a secondary look (eg. the lines are diagonals).

:four_leaf_clover:

r-a-v-a-s, 2015-07-29:
One look could be lines coming from each corner that terminate on the snap point.
Just because I feel like sharing, here is a related theorem: https://en.wikipedia.org/wiki/British_flag_theorem

atorrey, 2015-07-29:
I don't know what the best approach would be, though obviously making it 'automagical' would be nice...

I'm not a big user of hotkeys, I use enough different programs that I'd just as soon not remember what key combo does what in which program...  As a result it's slower but I tend to be much more of a mouse / menu based user...

I hadn't thought about changing the crosshair orientation to diagonals, but that could be a neat approach - except of course when trying to add a 45* diagonal.... <GRIN?>

ART

------------------
Arthur Torrey - <arthur_torrey@comcast.net>
-------------------

https://github.com/LibreCAD/LibreCAD/issues/550#issuecomment-126121520

atorrey, 2015-07-29:
I could see that being a very slick approach.  (I'm assuming by corner, you mean screen corners, it might be a problem figuring out what other points to use, and arguably less useful)

It would have minimal odds of significantly overlapping with either existing drawing features or what you are trying to add.  Even if it did overlap, that could almost always be fixed just by panning a bit in any direction....

Not completely sure it would work as nicely in practice as it sounds like it would in theory, but seems worth looking at...

As one option on how to make it easy to change the crosshair shape - I use KiCAD a good bit for doing schematics and wiring diagrams.  In the schematic drawing portion of the program "eeschema" they have a choice between an 'X' crosshair and dotted horizontal and vertical lines across the screen.  To switch between them, there is a clickable icon on one of the toolbars that toggles between the style...  Whichever mode you are in, the Icon shows a picture of the other one. (IIRC, there are also menu choices and hotkeys)

ART

------------------
Arthur Torrey - <arthur_torrey@comcast.net>
-------------------

https://github.com/LibreCAD/LibreCAD/issues/550#issuecomment-126126474

dxli, 2015-08-21:
please test with commit: fba9948b6fe2

atorrey, 2015-08-23:
I just downloaded and installed the version below - per the wiki instructions on how to install the latest daily

Version: master
SCM Revision: 2.1.0
Compiler: GNU GCC 4.8.4
Qt Version: 4.8.6
Compiled on: Aug 20 2015

Does this include the commit you mentioned?

I played with it some, and seems like it did pretty well, even when I tried to set the layer to use the same color and dash pattern as the crosshair, it still gave me enough contrast to see where both the existing features and what I was adding were....

Thanks!

ART

------------------
Arthur Torrey - <arthur_torrey@comcast.net>
-------------------

https://github.com/LibreCAD/LibreCAD/issues/550#issuecomment-133473954

dxli, 2015-08-23:
yes, this experimental feature is in the master branch (to become version 2.1) and in the latest v2.0.8 release.

We will see whether there's any regression due to this tweaking

atorrey, 2015-08-24:
Great, hope there are no other problems, as it seems to do well at solving the contrast problem.

ART

------------------
Arthur Torrey - <arthur_torrey@comcast.net>
-------------------

https://github.com/LibreCAD/LibreCAD/issues/550#issuecomment-133783460

r-a-v-a-s, 2015-09-11:
The contrast doesn't help when using the horizontal & vertical line tools.
Here is the alternative "crosshair" I suggested: https://youtu.be/E82N5OkPViA
Terrible video quality I know... I'm still trying to figure out [OBS](https://obsproject.com/).

Out of curiosity, is there a benefit to using the normal crosshair?

r-a-v-a-s, 2015-09-14:
Three ideas on how to implement it:

- replace the current crosshair
- create a toggle hotkey
- have the orthogonal tools set the crosshair

atorrey, 2015-09-15:
I have been swamped w/ other projects lately, so haven't had much time to play with it, but when I tried the contrast in the version right after you fixed it, (see previous messages for which one that was) it seemed to work pretty well even when using the horizontal and vertical tools that I tried.  I even tried doing a few 'dirty tricks' like setting the line color to be the same as the crosshair, and it still gave me a contrast...  This is on Kubuntu Linux, and a laptop screen if that matters...

The Youtube link is broken (removed by user error) so I can't see that, but IIRC you were basically suggesting making the crosshair a diagonal when using the H/V line tools, which seems like it would work...

The only real benefit I see to using the 'normal' crosshair is consistency in appearance, and that it sometimes can act as a 'guide' when moving in on a desired point - but I can live without those...

You might also look at the way they do the crosshair in the KiCAD 'eeschema' program - they have a choice of either a small thick crosshair with very short arms (think typical mouse cursor size) or a full screen width horizontal / vertical line that is a little thinner than the usual drawing elements, with a screen edge button to swap between them...

ART

------------------
Arthur Torrey - <arthur_torrey@comcast.net>
-------------------

https://github.com/LibreCAD/LibreCAD/issues/550#issuecomment-139687802

r-a-v-a-s, 2015-09-15:
I'm referring to the fact that the crosshair hides the preview objects, 
and there is no affect from the update by dxli.

I replaced the link when I edited the comment. https://youtu.be/0v9TS7IyvKM

atorrey, 2015-09-15:
What I found was that while the crosshair was on top of the preview objects, it didn't completely obscure them as the object caused that part of the crosshair to be a different color....  It was sometimes a bit of a challenge to totally see them, but definitely possible, and a big improvement over the previous version.

This time I was able to see the YouTube, and it was something that would work, though I found it a bit disconcerting...  I was thinking more of a crosshair that was a constant 45* from the cursor point, instead of 90* (X instead of + ) not the 'spider-web' from the corners...  However what you have there certainly works, and I could get used to it...

ART

------------------
Arthur Torrey - <arthur_torrey@comcast.net>
-------------------

https://github.com/LibreCAD/LibreCAD/issues/550#issuecomment-140550947

Osndok, 2015-11-09:
There is a cross-hairs change you might be interested in https://github.com/Osndok/LibreCAD/commit/9c610bcc2008d0ecdf85e610210526f8d491f604 I'm going to (attempt to) attach a screenshot so you can see it.
[image]

r-a-v-a-s, 2016-03-15:
@atorrey You can use the 'Spiderweb' or even 'Isometric' snap indicator. As a side note, you can also change the snap indicator's color now.

closed 2016-03-15 by r-a-v-a-s