LibreCAD/LibreCAD #551: Error compiling on Centos

jacusiel, 2015-08-02:
/usr/lib64/qt4/bin/uic ui/forms/qg_dlgoptionsmakercam.ui -o ../../generated/librecad/ui/ui_qg_dlgoptionsmakercam.h
Hello, I get the following error when compilimgo on Centos. Please tell me what aditional info you need. Thanks

uic: Error in line 46, column 34 : Unexpected attribute alignment
File 'ui/forms/qg_dlgoptionsmakercam.ui' is not valid
make[2]: *** [../../generated/librecad/ui/ui_qg_dlgoptionsmakercam.h] Error 1
make[2]: Leaving directory `/root/lintex/LibreCAD/librecad/src'
make[1]: *** [sub-src-make_default] Error 2
make[1]: Leaving directory `/root/lintex/LibreCAD/librecad'
make: *** [sub-librecad-make_default-ordered] Error 2

dxli, 2015-08-03:
can you test building with qt5?

jacusiel, 2015-08-03:
Hello, thanks for your reply. I'll try to install qt5 and compile.

closed 2016-09-07 by r-a-v-a-s