LibreCAD/LibreCAD #553: export pdf creates blank page

r-a-v-a-s, 2015-08-09:
Export → Export as PDF
always produces a PDF that is a single blank white page.

dxli, 2015-08-22:
can not reproduce this bug

bursi, 2015-09-02:
I experienced this problem as well, and in my case, it was caused by a scaling issue. 

You should check the dimensions and units of your drawing as well as the paper size. 
(both can be set in "Edit > Current Drawing Preferences")

If you have a drawing which is a few meters wide and you have an A4 paper selected, you'll only get a small cut of it on your pdf - which may very well be empty. 

To fix it, click "File > Print Preview" and then choose an appropriate scale. Then directly click "File > Export > Export As pdf". it is essential that you do so directly without closing the print preview. 

This worked for me, the drawing in the pdf had the same scale as in the print preview.

r-a-v-a-s, 2015-09-02:
Thanks for the information. I was able to create a non-blank pdf.

r-a-v-a-s, 2015-09-02:
I'm wondering if a dialog to choose the paper size would be appropriate.
https://en.wikipedia.org/wiki/Principle_of_least_astonishment

dxli, 2015-09-02:
we can modify the graphicview part to detect blank paper for printing, and even the detailed information:

like empty page, which covers only 0.01% of the drawing range.

r-a-v-a-s, 2015-09-03:
That seems viable.

kspartin, 2016-04-12:
Here is a drawing that I cannot export as PDF. I have it set for Arch E. I select a 1:10 scale and Export-> Export as PDF. This worked in 2.7 without a problem. This drawing was originally drawn in 2.7.
[Supports for Tank Access Platform - Sht 2.zip](https://github.com/LibreCAD/LibreCAD/files/215609/Supports.for.Tank.Access.Platform.-.Sht.2.zip)

kspartin, 2016-04-12:
2.0.8 and 2.0.9 have this problem. 2.0.7.1 works perfectly.

kspartin, 2016-04-15:
Basically, what I have figured out is that the english (us) paper sizes don't work. The metric A sizes seem to work fine.

r-a-v-a-s, 2016-04-15:
I confirm that it creates a blank PDF using the latest code.
Something else to note is that Arche E is described as 35.984 X 47.992
instead of 36 x 48.

r-a-v-a-s, 2016-04-15:
@kspartin  I just added support for the Arch paper sizes.
I'm not sure how it worked for you in 2.0.7...
the code was equally broken in that version.

DarrellDuane, 2018-05-21:
I couldn't get it to work whether printing to PDF or printing directly to my printer.  Eventually I exported it as a .PNG Image file and printed the image.

retiform, 2018-09-01:
Issue occurred for me as well.

ceremcem, 2018-09-10:
I'm also running 2.1.3 and my current workaround is 

1. Print Preview 
2. Set scale to 1:1 
3. Center page 
4. Print 
5. Select "Write to PDF" as printer

Only problem is that output is not 1:1, it's like 1:0.98 (scaled down).

sotitrox, 2019-08-23:
I have the same problem, but I modify some parameters of the layer, because, if you see more close the page exported its not a blank page, then I only modified the weight of the line, and **then I went to the print preview window** to export to pdf, works for me.

eshazen, 2020-08-06:
I am still getting blank pages in 2.1.3.  I've got a drawing in inches, and tried both the US letter size and custom size 8.5 x 11.  All produce blank pages.

Jontepello, 2020-11-24:
I solved it by editing my "Layer List" in the right sidebar. My added layers automaticaly became 0.0mm wide. By changing the value in that column to 0.05mm solved the export problem for me.
Using 2.1.3 Linux version.

DuncanLithgow, 2021-01-29:
I can't reproduce the 'blank page pdf' issue in 2.2rc2 with the file @kspartin posted. Anyone seeing this in that version?

fa201, 2021-01-30:
@DuncanLithgow I confirm the PDF shows the drawing with the following build:
Version: 2.2.0-rc2
Compiler: GNU GCC 7.5.0
Compiled on: Jan 10 2021
Qt Version: 5.9.5
Boost Version: 1.65.1
System: Ubuntu 18.04.5 LTS

[try.pdf](https://github.com/LibreCAD/LibreCAD/files/5898766/try.pdf)

closed 2021-01-30 by lordofbikes

Nodraak, 2021-02-25:
For anyone reading this, the issue was not the paper format in drawing preferences, but the width of the line:

* Right click on the layer
* Edit layer attributes
* Width
* I selected 0.25 mm because I like round metric numbers and 0.10 was not available. YMMV.

Lonniebiz, 2021-05-28:
The export to PDF feature, should (by default), automate all steps necessary to produce a PDF document that "scales to fit". It should even auto-choose portrait or landscape based on the aspect ratio of the drawing itself, so that the drawing is maximum size on the page of the PDF document.

Our staff, needs no other feature of LibreCAD than its **Export to PDF** feature! We don't edit any of the dmg or dxf files we receive from our customers. Most of our customers send us PDFs to begin with. We just need to know the measurements of the pieces for selling the raw material that the CAD requires.

Occasionally, the customer will send us a dmg or dxf file. Instead of asking the customer to re-send us a PDF, we were hoping we could just install LibreCAD onto each workstation and exclusively use it for its "Export to PDF" capability.

Our users aren't technical and need a tool that automatically does what I've mentioned above. I hope the 2.2.0-rc2 version, mentioned by @fa201 , has solved this issue to the degree I've mentioned above.

There are online tools that will convert dxf to pdf, but we value our customer's privacy and don't want to entrust their CADs with online 3rd parties.

When I saw the print preview in LibreCAD, it made me very excited that LibreCAD was exactly what we needed. As @eshazen said, that part looked great, but is was unfortunately not what ended up on the PDF!

I cannot burden my users with scaling and paper selection work-arounds. The PDF converter I use, needs to automate these things.

Take a look at, [DraftSight](https://www.draftsight.com/) "Export to PDF" feature, it was the best by default I've seen so far in my testing. I only mention as an example to model after. It is apparently doing the things I'm requesting in LibreCAD. However, we can't afford to put DraftSight onto every workstation (not open source / Free). So we're hoping LibreCAD will eventually gets this feature to be easy enough for our staff's convenience requirements.

cdnthought, 2022-03-06:
I can confirm that printing is really buggy, especially when using a PDF Printer (Microsoft Built-in or CutePDF tested) or the built-in export to PDF function.  The issue seems to rear its head when using US sizes like "Tabloid", but using EU sizing like "A3" works better.  Overall, I'd say there needs to be more work on making printing function better and more like other programs - Print Preview should allow you to change orientation, paper size, and printable area with options all in the forefront, not hidden in a menu.  Or if it is simply a "preview" then a Print Setup option needs to be added that focuses on those options.  Then, and this is the more important part, it needs to play nice with MS Print to PDF or work with the built-in export to PDF function, and it needs to work for all sizes listed.  At least that's where I'd want my dono funding to go towards.  It's frustrating to work on my house design for 3 months and now be unable to print or export to PDF without blank pages, or "printing complete" messages with no "save as" box, or portrait prints that cut off half the drawing even though print preview showed landscape.  Makes this program useless until I found the work around to use the A3 size as a proxy for Tabloid.  All other programs on my PC work well with CutePDF and MS Print to PDF except Librecad, a sure sign that there's something incomplete under the skin.

Jo-le-Papy, 2022-05-07:
> For anyone reading this, the issue was not the paper format in drawing preferences, but the width of the line:
> 
>     * Right click on the layer
> 
>     * Edit layer attributes
> 
>     * Width
> 
>     * I selected 0.25 mm because I like round metric numbers and 0.10 was not available. YMMV.

Yes, it work! this is the solution. Than'k