LibreCAD/LibreCAD #554: location bar sliders WAY out of proportion to drawing size

atorrey, 2015-08-23:
If I have a part drawn, I feel that the slider bars that can be used for panning should be sized such that one can move around the actual drawing area, but not be so small that it is easier to lose the drawing than to keep it on the screen...

Attached is what I get when I auto-zoom the drawing, which as I understand it should include everything in the file, so there is nothing of interest elsewhere in the image...  Note the TINY sliders (emphasis added in photo)....  If I zoom in they stay about the same size.  If I zoom out, the bars get bigger slowly - by the time they are a third of the bar, the image is so small I can't tell it from a grid dot, which is hardly useful...

In other graphics type applications I use, the sliders are sized proportionately to the image, and are useful for panning around it, and I use them a lot....  Here they are so small that even tiny movements of the sliders give HUGE movements of the drawing, often sending me off into unknown space....

It would be better if the sliders were proportional to the drawing - so that the autozoom drawing slider was 100% of the bar size, and then got smaller as you zoomed in....  If needing to expand the drawing, let them stay at 100% as you zoom out, and then shrink as you go back in, but have auto-zoom reset them to 100%
[image]

(drawing was an existing one, opened in 
Version: master
SCM Revision: 2.1.0
Compiler: GNU GCC 4.8.4
Qt Version: 4.8.6
Compiled on: Aug 20 2015
which I think is the latest....)

atorrey, 2015-08-23:
[image]

Drawing didn't upload last time - not sure why...

dxli, 2015-08-24:
there was a bug in deciding scroll ranges.

the actual bug:

range is in GUI space, then:

toGuiDX(getWidth()) should be getWidth() only, since getWidth() is already in GUI space.

closed 2015-08-24 by dxli