LibreCAD/LibreCAD #555: 2.0.8 fails to build on Fedora 22 / 32bit

prokoudine, 2015-08-25:
I get this error message when building 2.0.8 with both Qt4 and Qt5:

../../generated/librecad/obj/rs_filterdxfrw.o: In function `DRW_AppId::DRW_AppId()':
/home/alex/soft/cad/librecad/LibreCAD-2.0.8/librecad/src/../../libraries/libdxfrw/src/drw_objects.h:484: undefined reference to `vtable for DRW_AppId'
../../generated/librecad/obj/rs_filterdxfrw.o: In function `RS_../../generated/librecad/obj/rs_filterdxfrw.o: In function `DRW_AppId::DRW_AppId()':
/home/alex/soft/cad/librecad/LibreCAD-2.0.8/librecad/src/../../libraries/libdxfrw/src/drw_objects.h:484: undefined reference to `vtable for DRW_AppId'
../../generated/librecad/obj/rs_filterdxfrw.o: In function `RS_FilterDXFRW::writeAppId()':
/home/alex/soft/cad/librecad/LibreCAD-2.0.8/librecad/src/lib/filters/rs_filterdxfrw.cpp:1975: undefined reference to `dxfRW::writeAppId(DRW_AppId*)'
../../generated/librecad/obj/rs_filterdxfrw.o: In function `DRW_AppId::~DRW_AppId()':
/home/alex/soft/cad/librecad/LibreCAD-2.0.8/librecad/src/../../libraries/libdxfrw/src/drw_objects.h:481: undefined reference to `vtable for DRW_AppId'
../../generated/librecad/obj/rs_filterdxfrw.o: In function `RS_FilterDXFRW::fileImport(RS_Graphic&, QString const&, RS2::FormatType)':
/home/alex/soft/cad/librecad/LibreCAD-2.0.8/librecad/src/lib/filters/rs_filterdxfrw.cpp:130: undefined reference to `dwgR::dwgR(char const*)'
/home/alex/soft/cad/librecad/LibreCAD-2.0.8/librecad/src/lib/filters/rs_filterdxfrw.cpp:134: undefined reference to `dwgR::read(DRW_Interface*, bool)'
/home/alex/soft/cad/librecad/LibreCAD-2.0.8/librecad/src/lib/filters/rs_filterdxfrw.cpp:142: undefined reference to `dwgR::~dwgR()'
/home/alex/soft/cad/librecad/LibreCAD-2.0.8/librecad/src/lib/filters/rs_filterdxfrw.cpp:130: undefined reference to `dwgR::~dwgR()'
/home/alex/soft/cad/librecad/LibreCAD-2.0.8/librecad/src/lib/filters/rs_filterdxfrw.cpp:133: undefined reference to `dwgR::setDebug(DRW::DBG_LEVEL)'
/home/alex/soft/cad/librecad/LibreCAD-2.0.8/librecad/src/lib/filters/rs_filterdxfrw.cpp:130: undefined reference to `dwgR::~dwgR()'
../../generated/librecad/obj/rs_filterdxfrw.o:(.rodata._ZTV13DRW_DimLinear[_ZTV13DRW_DimLinear]+0x14): undefined reference to `DRW_DimAligned::parseDwg(DRW::Version, dwgBuffer*, unsigned int)'
collect2: error: ld returned 1 exit status
Makefile:1255: recipe for target '../../unix/librecad' failed
make[2]: *** [../../unix/librecad] Error 1
make[2]: Leaving directory '/home/alex/soft/cad/librecad/LibreCAD-2.0.8/librecad/src'
Makefile:39: recipe for target 'sub-src-make_default' failed
make[1]: *** [sub-src-make_default] Error 2
make[1]: Leaving directory '/home/alex/soft/cad/librecad/LibreCAD-2.0.8/librecad'
Makefile:78: recipe for target 'sub-librecad-make_default-ordered' failed
make: *** [sub-librecad-make_default-ordered] Error 2
FilterDXFRW::writeAppId()':
/home/alex/soft/cad/librecad/LibreCAD-2.0.8/librecad/src/lib/filters/rs_filterdxfrw.cpp:1975: undefined reference to `dxfRW::writeAppId(DRW_AppId*)'
../../generated/librecad/obj/rs_filterdxfrw.o: In function `DRW_AppId::~DRW_AppId()':
/home/alex/soft/cad/librecad/LibreCAD-2.0.8/librecad/src/../../libraries/libdxfrw/src/drw_objects.h:481: undefined reference to `vtable for DRW_AppId'
../../generated/librecad/obj/rs_filterdxfrw.o: In function `RS_FilterDXFRW::fileImport(RS_Graphic&, QString const&, RS2::FormatType)':
/home/alex/soft/cad/librecad/LibreCAD-2.0.8/librecad/src/lib/filters/rs_filterdxfrw.cpp:130: undefined reference to `dwgR::dwgR(char const*)'
/home/alex/soft/cad/librecad/LibreCAD-2.0.8/librecad/src/lib/filters/rs_filterdxfrw.cpp:134: undefined reference to `dwgR::read(DRW_Interface*, bool)'
/home/alex/soft/cad/librecad/LibreCAD-2.0.8/librecad/src/lib/filters/rs_filterdxfrw.cpp:142: undefined reference to `dwgR::~dwgR()'
/home/alex/soft/cad/librecad/LibreCAD-2.0.8/librecad/src/lib/filters/rs_filterdxfrw.cpp:130: undefined reference to `dwgR::~dwgR()'
/home/alex/soft/cad/librecad/LibreCAD-2.0.8/librecad/src/lib/filters/rs_filterdxfrw.cpp:133: undefined reference to `dwgR::setDebug(DRW::DBG_LEVEL)'
/home/alex/soft/cad/librecad/LibreCAD-2.0.8/librecad/src/lib/filters/rs_filterdxfrw.cpp:130: undefined reference to `dwgR::~dwgR()'
../../generated/librecad/obj/rs_filterdxfrw.o:(.rodata._ZTV13DRW_DimLinear[_ZTV13DRW_DimLinear]+0x14): undefined reference to `DRW_DimAligned::parseDwg(DRW::Version, dwgBuffer*, unsigned int)'
collect2: error: ld returned 1 exit status
Makefile:1255: recipe for target '../../unix/librecad' failed
make[2]: *** [../../unix/librecad] Error 1
make[2]: Leaving directory '/home/alex/soft/cad/librecad/LibreCAD-2.0.8/librecad/src'
Makefile:39: recipe for target 'sub-src-make_default' failed
make[1]: *** [sub-src-make_default] Error 2
make[1]: Leaving directory '/home/alex/soft/cad/librecad/LibreCAD-2.0.8/librecad'
Makefile:78: recipe for target 'sub-librecad-make_default-ordered' failed
make: *** [sub-librecad-make_default-ordered] Error 2

Suggestions?

r-a-v-a-s, 2015-08-26:
How are you building?
I once solved the problem `...undefined reference to vtable for...` in Qt Creator by:

1. Build → Run qmake
2. Build → Rebuild All

prokoudine, 2015-08-26:
qmake-qt4 librecad.pro
make

r-a-v-a-s, 2015-08-26:
You could try `qmake -project` in the directory that has the pro file.

> In this mode, qmake will search the current directory for files with known extensions (.h, .cpp, .ui, etc.) and produce a .pro file that lists these files.

http://www.informit.com/articles/article.aspx?p=1405563&seqNum=2

r-a-v-a-s, 2015-08-26:
Although in theory you received the proper .pro files with the latest source...

prokoudine, 2015-08-27:
Unfortunately that way the build breaks even earlier.

dxli, 2015-08-27:
Please test it from a clean folder, something like:

```Shell
pushd $(mktemp -d)
unzip /path/to/LibreCAD-2.0.8.zip
cd LibreCAD-2.0.8/
qmake -r  2>&1 > ../build.log
make -j3 2>&1 >> ../build.log
```
If it still fails, please share your build log file.

cd tmp

Rallaz, 2015-08-28:
Hi Alexander,
Check with these lines:

    qmake-qt4 -makefile -nocache librecad.pro
    make clean
    make

You can download 2.0.8 binary packages for fedora 20, 21 & 22 (32 & 64 bits) from
http://download.opensuse.org/repositories/home:/Rallaz/

The minimum required packages are:
libdxfrw0-0.6.1-1.1 & librecad-2.0.8-1.1

prokoudine, 2015-08-28:
Hi,

@dxli, here's the log:

http://libregraphicsworld.org/files/librecad-build.log.bz2

@Rallaz, awesome! I'll try your builds as soon as I'm back at home ;)

dxli, 2015-09-01:
the original building error indicates an incompatible libdxfrw package version.

Could you try to uninstall the libdxfrw package and try to build from source again?

closed 2016-09-07 by r-a-v-a-s