LibreCAD/LibreCAD #560: Copy-paste from a DXF to another makes LibreCAD freeze

Rupicapra-rupicapra, 2015-09-03:
Hello,
I noticed a weird bug in 2.0.8 (I never got it under 2.0.2) : when I copy-paste content from a dxf to another (this content contains polylines, lines, block[polylines,hatching]), LibreCAD freezes and one of my CPU cores gets monopolized at 100 % by LC (exiting program window does not frees the core, I have to kill LC on command line -> if I re-start LC and it freezes again, 2 cores are fully used, etc.). Waiting for some minutes does not solves the freezing.
I also noticed the same problem on Windows 7 with LC 2.0.8.
I tried to run LC in debug mode, but I didn’t have any information when it freezed.

Freezing seems to be linked to some elements I want to copy : when I remove some lines (from a particular layer) or block, it does not freezes…

dxli, 2015-09-03:
please upload a testcase dxf

Rupicapra-rupicapra, 2015-09-04:
I'll prepare two testcase dxfs.
I noticed that I also have this bug with some blocks (stored in my personal library) : select in library explorer -> insert -> insert once, no problem -> insert twice, no problem -> clicking to insert it a third time -> instantly freezes like described above.

r-a-v-a-s, 2016-09-07:
Some work has been done on this.
Please try the latest commit.

fa201, 2020-04-14:
@lordofbikes 
Should we close this issue ?