LibreCAD/LibreCAD #562: Shortcuts for drawing order

Rupicapra-rupicapra, 2015-09-07:
Hello,
I want to set shortcuts for drawing order modifiers (menu Edit > Draw order), and I would like to have your opinion on the best shortcuts to use.

For 'Move to Top' & 'Move to Bottom', I would like to use these shortcuts : Ctrl + PageUp & Shift + PageDown.
For 'Lower after entity' & 'Raise over entity', I was thinking about Ctrl + up & Ctrl + down.

Another way would be using inkscape-like shortcuts (do not use modifier keys) :
 - 'Move to top' -> 'Start';
 - 'Move to bottom' -> 'End';
 - 'Lower' -> PageDown;
 - 'Raise' -> PageUp.
What do you thing about these possibilities ? Would you prefer other shortcuts ?

Thanks,

Rupi

severoraz, 2015-09-07:
I think this is a good idea

dxli, 2015-09-08:
merged #564

dxli, 2015-09-08:
Thanks a lot to jerry: @Rupicapra-rupicapra

closed 2015-09-09 by dxli