LibreCAD/LibreCAD #563: Zoom with touchpad has exaggerated zoom factor

severoraz, 2015-09-07:
I am using LibreCAD 2.0.8 on a Dell XPS M1530 with Arch Linux x86_64 and GNOME 3.16, and when I zoom using the touchpad I can't control how much to zoom, and I easily reach zooming limits and LibreCAD stalls for a bit when this happens.

(I once talked about this to a dev at freenode.net#librecad and he suggested this was due to a recent change to support high resolution touchpads, which made sense.)

r-a-v-a-s, 2015-09-09:
How do you zoom with the touchpad?

severoraz, 2015-09-09:
Sorry, I really should've explained that haha, my touchpad has an integrated scrollbar along its right side over which I can swipe my finger to scroll.

dxli, 2015-09-09:
right now, the default zooming factor is by 1.25

I do feel we should make zooming slower, for example, switching to 1.1 or 1.05. However, that's not a real fix

Rupicapra-rupicapra, 2015-09-10:
Maybe we can have a look to [QTouchEvent class](http://doc.qt.io/qt-5/qtouchevent.html) (manages touch events -> touchpad, but also touch screen, tablet…) to specifically handle zooming action with a touch device (and applying a lower zoom factor). It would keep 1.25 factor for mouse zooming (a better option could be to let user define a zoom factor in general properties window -> I can have a look at this in the next few days if you find it interesting).

severoraz, 2015-09-10:
Awesome, it pleases me very much to see quick response in this project. Right now I was building with the zoom factor changed to 1.1 to see how this came out, but I will look into this QTouchEvent class

dxli, 2015-09-10:
talking about touch screen, it's always interesting to see whether we can run LibreCAD on android.

I got the impression, GPL license is not compatible with Apple, so we can not do iOS for v2, but v3 should be fine

rubdos, 2016-04-28:
Hijacking this issue a little, and I'm not sure whether that's right or not. I (Arch 86_64, libinput) upgraded librecad from 2.0.9 to 2.0.10 some days ago. My first user experience was on 2.0.9 some weeks ago and I loved it.

Since 2.0.10 though, my scrolling behavior is uncontrollable, as @severoraz says. Every tiny bit of scrolling triggers an immense amount of zoom. Before, scrolling was for navigating the workspace.

I'm on X.org and a Lenovo Thinkpad X250, didn't test on Wayland.

My `cat /var/log/pacman.log | grep librecad`, fyi:

> [2016-01-27 11:56] [ALPM] installed librecad (2.0.9-1)
> [2016-04-20 14:03] [ALPM] upgraded librecad (2.0.9-1 -> 2.0.10-1)
> [2016-04-27 21:29] [ALPM] upgraded librecad (2.0.10-1 -> 2.0.10-2)
> [2016-04-28 09:25] [PACMAN] Running 'pacman -U librecad-2.0.9-1-x86_64.pkg.tar.xz'
> [2016-04-28 09:25] [ALPM] downgraded librecad (2.0.10-2 -> 2.0.9-1)

As this is a regression, that probably didn't happen on librecad 2.0.8 (although I could test that), I'm not sure whether it's good that I hijack this thread. If it's not the case, tell me and I'll open another issue.

Thank you!

severoraz, 2016-04-28:
[4/28/2016 1:50] Ruben De Smet:
> Every tiny bit of scrolling triggers an immense amount of zoom. 
> Before, scrolling was for navigating the workspace.

This sounds just like the issue to me, so don't worry about hijacking 
the thread :)

I have not tested this in a while though so I wouldn't know at this time 
how LibreCAD behaves with my touchpad.

r-a-v-a-s, 2016-04-28:
@rubdos Please state your operating system and the version of Qt used to build LibreCAD.

r-a-v-a-s, 2016-04-28:
I'm guessing this is because of https://github.com/LibreCAD/LibreCAD/commit/4a8003e88c228322bc1a529985efccf3d1316502

rubdos, 2016-04-29:
I already mentioned Arch Linux x86_64.

I'm on kernel 4.6, qt5-base 5.6.0-4, qt4 4.8.7-8. I'm not sure which version (qt4 or qt5) librecad uses for building though. I'm just using the Arch Linux default build.

Thank you :)

r-a-v-a-s, 2016-04-29:
Look in the Help -> About window.

rubdos, 2016-04-29:
Qt 5.6.0

[image]

r-a-v-a-s, 2016-04-29:
Thanks!

Would you be willing to build the master branch?

If you already have Qt the only other thing you need is Boost.
Although, on openSUSE I also needed freeglut-devel.
https://github.com/LibreCAD/LibreCAD/wiki/Becoming-a-developer

Then in theory you can open librecad.pro in Qt Creator,
in the Projects section (left sidebar) uncheck "Shadow build"
and maybe add -j3 to the make arguments for faster compiling.

Then in LibreCAD: Options->Device Options
and set it to trackpad.

If that doesn't work you will have to play around with the relevant code:
https://github.com/LibreCAD/LibreCAD/commit/4a8003e88c228322bc1a529985efccf3d1316502

I didn't notice anything I can fix;
and I don't have a trackpad / touchpad to test with.

rubdos, 2016-04-29:
Wait what, there's an option for trackpads? Lemme test that first then.

If it doesn't work, I will play with the code and send in a patch if necessary.

r-a-v-a-s, 2016-04-29:
Not in 2.0.9, the master branch is 2.1.0-alpha.
https://github.com/LibreCAD/ChangeLogs/blob/master/developers/r-a-v-a-s/2-1/2-1-0/2-1-0.md

rubdos, 2016-04-29:
He, there's a `librecad-git` PKGBUILD. I'm installing that, will report back.

rubdos, 2016-06-11:
I installed 2.1.0 (it's in Arch repo's); after setting to trackpad, everything is awesome again!

severoraz, 2016-06-12:
I also have 2.1.0 and in Arch Linux zooming in and out with trackpad with trackpad device set, works excellent! I cannot test it with the computer I initially reported this bug with, since it is unserviceable now. Anyway, support for different devices sure seems like the solution for this bug =D

r-a-v-a-s, 2016-06-12:
Thanks for the feedback. 😎

closed 2016-06-12 by r-a-v-a-s