LibreCAD/LibreCAD #565: anti-aliasing

r-a-v-a-s, 2015-09-09:
I can add an option for anti-aliasing in the drawing area if that is agreeable.

dxli, 2015-09-09:
no reason not to play with it. I suppose there's some performance cost, but shouldn't be much.

severoraz, 2015-09-10:
I'm all in favor for this option!

r-a-v-a-s, 2015-09-10:
I'll get to this in a few days, after the gui update is polished.

What do you guys think:
- an option in application preferences
- or a toggle icon (like the grid toggle)

dxli, 2015-09-10:
I don't suppose this would be tweaked often, so within application preferences is fine

it's global and takes effect immediately, unlike many application preferences which requires restarting of the app to take effect.

severoraz, 2015-09-11:
I also favor the option within application preferences (if I have vote haha)

closed 2015-09-22 by r-a-v-a-s

severoraz, 2015-09-22:
In which build will we find this feature?

lordofbikes, 2015-09-22:
You will find it in the last night build installer for Windows: *LibreCAD-Installer-2.0.7-603-g00eb65b.exe* at https://sourceforge.net/projects/librecad/files/Windows/NightlyBuilds/master/.

If you build from source, pull latest changes and ensure that you are on the master branch.

severoraz, 2015-09-24:
I built it, looks great!