LibreCAD/LibreCAD #569: hide the titlebar

r-a-v-a-s, 2015-09-16:
I can add the ability to toggle the visibility of the titlebar. 
This will increase available vertical space.
I'm thinking F11 for the toggle.

atorrey, 2015-09-16:
If this is done, (and while it is something I don't think I would use, it sounds reasonable,) PLEASE make sure there is a way in a menu that CAN'T be hidden to bring it back....

I have been annoyed on more than a few occasions when I've accidentally hit a function key by accident (I'm both fat fingered and often drop the keyboard...) and done something to the display that had no obvious / easy way to reverse....

ART

------------------
Arthur Torrey - <arthur_torrey@comcast.net>
-------------------

https://github.com/LibreCAD/LibreCAD/issues/569

dxli, 2015-09-16:
@atorrey ,

yes, it should be in the "View" menu

r-a-v-a-s, 2015-09-26:
The [built-in fullscreen mode][1] works really well on Windows 7;
it even hides the taskbar. :sunglasses: 

[1]:http://doc.qt.io/qt-5/qwidget.html#showFullScreen

closed 2015-09-26 by r-a-v-a-s