LibreCAD/LibreCAD #570: command info

r-a-v-a-s, 2015-09-16:
We could add a HTML table to the internal help.
I imagine a table like the one I made for the wiki that also includes an icon column.

Another idea is to add at least one command to the action tooltips.

Another idea is to have command line output 
that lists the associated commands when you activate a tool via its icon.

r-a-v-a-s, 2015-11-11:
Now I'm thinking:
- use `setStatusTip()` to display the commands for an action
- create a help mode and use `setWhatsThis()` for a description of the tool

Harvie, 2015-11-11:
Would be nice if there was at least configurable option to show commands in menus and maybe live in commandline.
Also some 10-20 basic commands can be shown when i enter "help" command or "?" as it's what i usually do when i start exploring some commandline interface which is new for me.

r-a-v-a-s, 2015-11-11:
My proposal means that when you hover over an icon or menu entry
you will see the commands in the statusbar.

You can currently press tab to see all commands;
and you can press tab after entering a letter 
to see any commands that start with that letter.

There is also http://wiki.librecad.org/index.php/Commands
which I will update once the move to dokuwiki is complete.

r-a-v-a-s, 2016-01-30:
@Community
You will see the commands in the statusbar when you hover over the modify tools.
This is a little faster than waiting for a tooltip to popup.
Tell me what you think.

If anyone is interested in contributing by completing this project,
you simply need to look at the commands in rs_commands.cpp
and add them to lc_actionfactory.cpp (view my previous commit).

Leave a comment if you are interested; otherwise I will complete this in a few days.

r-a-v-a-s, 2016-02-08:
I've been noticing the status bar doesn't always update after choosing a modify action from the categories toolbar. In other words, the commands continue to be the only thing in the status bar.
I'll start working on tooltip modification instead.

r-a-v-a-s, 2016-02-12:
Tooltips do not appear over menus on OSX.

My latest idea is printing the information in the command line's output
when a command is triggered. Example: `Attributes : modifyattr, attr, ma`

fa201, 2020-04-14:
@lordofbikes 
Tool tip works. Menu **Help/On line/Commands** also works. Should we consider this issue closed ?

sand1024, 2026-09-26:
should be closed after PR commit with settings and cheatsheet generation