LibreCAD/LibreCAD #572: features page basicly is empty

abma, 2015-09-16:
i've just read about LibreCAD but the features page isn't really informative for people who never heard about it.

( http://librecad.org/cms/home/features.html )

- what file formats can it open
- what file format can be written to (...CAD file formats)?
- can formats exported (i.e. pdf, png, ...)?
- can special input devices be used?
- etcetc

wikipedia has some hints: https://en.wikipedia.org/wiki/LibreCAD

r-a-v-a-s, 2015-09-16:
I think the only people with access to the website are busy with other things.
Also, we happen to be in the processes of switching the wiki framework.
Therefore the only way for the community to work on this is the github wiki.
I started a page: https://github.com/LibreCAD/LibreCAD/wiki/Features
I encourage others to work on it, so I can work on code. :sunglasses:

r-a-v-a-s, 2015-09-16:
You can use markdown for the wiki.
https://guides.github.com/features/mastering-markdown/

r-a-v-a-s, 2015-10-28:
Everyone who can work on the website has seen this by now.
Github issues are for things that can be dealt with via Git.
You can give any type of feedback at:
http://forum.librecad.org/Suggestions-amp-Feedback-f5706791.html

closed 2015-10-28 by r-a-v-a-s