LibreCAD/LibreCAD #573: calibrating print output

SergeyShorin, 2015-09-21:
Let user choose and printing any part of their own drawing (squares, circles, mm or inch rulers, protractor etc) in 1:1 scale.
If drawing large that paper - cut off the excess by positioning a sheet of paper (as they doing this now in print previews - by dragging the mouse sheet of paper "under the drawing").
After printing user can measure drawing on paper (any part - line, arc, angle etc) and calculate height and width scale (or just enter "drawing" height and width and "measured on paper" height and width).
And then press "SaveCalibrationForThisPrinter" button.
Calibrate data useful stored in local file (not with drawing file) where listed all printers (local or network, installed or connected for this computer) for each printer independently and separated.

* may be use internal paper measured template - user print him, measure specified (by developer) lines and put measured value in specified input box.

hgruen, 2016-01-07:
Thank you for this request. I would like this feature very much!
Can you mark your post as "request".?

aornt, 2018-03-07:
Yes, I would like that option too.
Trying to make a punch template for making holes in a frontpanel and the holes shift 1 mm on the printed page width.